Ultimaker/Cura #20579: Import multipart models position changes

JaVelin-X, 2025-05-05:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Biqu B1 ABL

### Reproduction steps

I have models to print where I need to have sections with no top or bottom layers to expose the infill. I create the model in CAD and import as a .3MF (or STL, OBJ it doesn't mater) in version 5.3.0 the model show up on screen in the correct positions but not on the build plate. it is possible to move them with out disturbing their relative positions so t's usable. in 5.8.0 and 5.10.0 the position of the models show up on the build plate but rearranged  so the positions are disturbed and are therefore unusable. All the drop to plate options are turned off that I can find and the forums say this ia bug thats existed for a while

### Actual results

Part position at import makes this version unusable for what I need

### Expected results

I expect parts to remain in the position they were in CAD when auto arrange or drop to plate are disabled

### Add your .zip and screenshots here ⬇️

[For print AP06-23H-46B v1.zip](https://github.com/user-attachments/files/20038306/For.print.AP06-23H-46B.v1.zip)

GregValiant, 2025-05-05:
Thanks for the report.
I see two models in the project file.  If I use "Select all Models" and then "Merge models" they go together, but there is no way for me to tell if the assembly is correct.  One model is inside the other.
AutoArrange will take each model and adjust it's position to where it thinks the best position is to print in relation to the other models.  It moves them around and may rotate them.
Drop to Plate will move each model so that the bottom-most point of each model is on the built plate.  It pays no attention to relationship between one part and another, just set each to Z=0.

I don't see a bug here.

JaVelin-X, 2025-05-05:
Hi. Thanks for your response

The problem with this approach is it makes the model one part. I'm trying
to use a per model setting so one of the parts is a "cutting mesh".

Curiously. if I undo that merge command twice. so (Select all models/Merge
Models/Undo/Undo)  The models revert to a state I experienced in 5.3.0
which is usable heh.  instead of really undoing what I did to get there.
Useful for me but not expected behaviour. When I import a multi part model
I want it to import just like that. I don't even care if it's now over the
build plate.

GregValiant, 2025-05-05:
It looks to me that you just needed to figure out how to do it.
Merging the models jumps them into place and then "ungrouping" them allows you to make the inner piece a cutting mesh.
[image]

The normal "Merge Models" operation is to print each part with a different extruder.  What you are trying to do is outside that normal situation but it does appear there is a workaround for it.
I still don't see a bug here.  You have a niche process you are trying to get Cura to do.  Sometimes that takes some creative thinking.

I'm inclined to remove the bug label and close this.  Let me know if you think that is premature.

JaVelin-X, 2025-05-05:
Nope all good i can make it work now. The only thing is the undo reverses
to a place that it never was

GregValiant, 2025-05-05:
Because 3mf files (generally) have the model location within them, I don't know how far along that information stays with the model.  Cura does something with the models and (for example) when you make a copy of a model on the Cura build plate, it an become detached from the original file.  You might be able to reload the original models, but not the copies.  There is some behavior there that has changed over the years.

I guess we're good here so I'll go ahead and close this.

closed 2025-05-05 by GregValiant