Ultimaker/Cura #20581: It showed unexpected error while slicing the part

Rutvik292, 2025-05-05:
### Cura Version

5.6.0

### Operating System

Window 11

### Printer

Elegoo Neptune 4 MAX

### Name abnormal settings

no

### Describe model location

centre 

### Describe your model

Tree trunk

### Add your .zip here ⬇️

Shows continuous Issues.

Asterchades, 2025-05-06:
Without the requested project file there's very little that can be done with this. Unless someone can replicate the issue it's simply not possible to diagnose further.

That said, every new version of Cura incorporates fixes for model slicing failures, with 5.7 in particular fixing an issue responsible for the lion's share of faults. Before submitting a project file I would highly encourage you to update your installation and try again - the latest release version of Cura is 5.10.0, released March 25.

GregValiant, 2025-05-08:
@Rutvik292 are you using Elegoo Cura (current version 5.6) or UltiMaker Cura (current version 5.10.0)?
If it is Elegoo Cura then you would need to post a bug report on their site.

Since this is either in regards to "Elegoo Cura" (wrong software), or an out-of-date version of UM Cura (5.6.0) - I'll go ahead and close this.

closed 2025-05-08 by GregValiant