Ultimaker/Cura #20585: Cura 5.10 won't open under windows 10 with dual monitors/video cards

papagino007, 2025-05-08:
### Cura Version

5.10.0

### Operating System

Windows 10 Pro version 22H2

### Printer

Tronxy XY-2 Pro

### Reproduction steps

Unable to open Cura, no splash screen open after clicking the shortcut, nothing happens. Task Manager doesn't show Ultimaker-cura opened in processes.

- My PC is using Windows 10 Pro version 22H2 with an Intel(R) Core(TM)2 Quad CPU Q9550, O.C. at 3.60 GHz.
- I have two NVIDIA GeForce GTX 950 video cards running 2 monitors, one monitor connected to each video cards. Using latest driver Version 576.28, Released Wed Apr 30, 2025.

I tried connecting only one monitor as suggested in forums, but still the same problem.
I tried using the two monitors in Show only on 1, duplicated and extended mode, same problem.
I tried running Cura "Run as Administrator" mode, same problem.
I tried uninstalling Cura and reinstalling in C:\ directory instead of default directory, same problem.
I tried installing older versions of Cura, same problem.

Any idea what is causing Cura to not open at all with no errors showing?

Any help would be appreciated.
Cheers

### Actual results

Cura doesn't open, no errors, no splash screen.

### Expected results

Well, I wish I was able to use cura to do slicing...

### Add your .zip and screenshots here ⬇️

N/A

Asterchades, 2025-05-09:
Open the config file (should be *%appdata%\cura\5.10\cura.cfg*) in a text editor like Notepad and look for the lines starting with "window_". There should be five such lines at the bottom of the [general] section. Delete these lines, save the file, and try launching Cura again. This should reset the window's size, location, and state.

If this works you can just keep going as is, or open up the *Preferences* and un-check *Restore window position on start* to stop it using those lines in future.

Asterchades, 2025-05-27:
This is going to require a log file for further diagnostics. You should be able to find it here: *%appdata%\cura\5.10\cura.log*

If it's completely absent that indicates it's not even being permitted to launch, which (to me, at least) would point more towards outside interference. You might be able to find something in either the OS *Event Viewer* or *Reliability Monitor* giving some detail (though probably not a lot) if this is the case.

papagino007, 2025-05-30:
Cura never starts, I get this from the Reliability Monitor:

Source
UltiMaker-Cura.exe

Summary
Stopped working

Date
‎5/‎30/‎2025 1:34 PM

Status
Report sent

Description
Faulting Application Path:	C:\Program Files\UltiMaker Cura 5.10.0\UltiMaker-Cura.exe

Problem signature
Problem Event Name:	APPCRASH
Application Name:	UltiMaker-Cura.exe
Application Version:	0.0.0.0
Application Timestamp:	67e12d6c
Fault Module Name:	nvoglv64.dll
Fault Module Version:	32.0.15.7628
Fault Module Timestamp:	680e5f6a
Exception Code:	c000001d
Exception Offset:	0000000000a04643
OS Version:	10.0.19045.2.0.0.256.48
Locale ID:	1033
Additional Information 1:	75bc
Additional Information 2:	75bcf92ccfb6d8512e63b68a1cc9df0b
Additional Information 3:	13bb
Additional Information 4:	13bbc3e67577738e1f06ea0dfad3e8c6

Extra information about the problem
Bucket ID:	bd71438039ee9db08a790cd0e5e75207 (1907570008517988871)

Asterchades, 2025-05-31:
Well, that's perhaps a start. Exception 0xc000001d means an invalid command has been issued while nvoglv64.dll is part of NVidia's OpenGL driver. This would suggest that either Cura is sending invalid commands to the DLL or that the DLL itself is sending invalid commands, though unfortunately it also means the possibility that something else is interfering that's being attributed to one or the other (eg the Nahimic audio service is known to cause this sort of problem with many versions).

My first instinct would be to reinstall both Cura itself and the display drivers. Unless you've done extensive profile configuration you may even consider using DDU to completely remove your graphics drivers on the way.

With that said, there are known complications with NVidia drivers at present. Unless you're using a 50-series (you indicate you're using a 900-series) it's generally better to avoid any and all drivers in the 570s - and even then you only use them for a 50-series as you've got no other choice. 566.36 are generally considered to be the last "good" drivers for the moment.

papagino007, 2025-05-31:
Funny how you mentioned using DDU to completely remove all traces of video driver as this is exactly what I did prior to my last message. I installed the latest 576.52 NVidia driver and reinstalled Cura and that didn't resolve the problem.
I tried running Cura after booting Windows in Safe Mode and got this error message: "Failed to probe OpenGL. Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers".
(I think this is normal as only the default Windows video driver was used in safe mode...)

I have 4 Sound Controller listed in Device Manager: 
- NVIDIA High Definition Audio (x2)
- SoundMAX integrated Digital HD Audio
- VF0610 Live! Cam Socialized HD

I disable the two NVIDIA High Definition Audio to see if this would be the cause, but no...

Cura used to run fine on this PC, but if I recalled correctly, it stops working either after I upgraded the video card drivers or Cura or after I installed a second NVIDIA GeForce GTX950 video card to drive my second monitor, prior, the two monitors were connected to only the one GTX950 card.

I do believe it is like you say related to OpenGL.

I will try removing my current latest driver and install version 566.36 as I have that file also.

I will post any positive results if this is the case...

papagino007, 2025-06-01:
No it wasn't successful...
Something bad happen when trying to update the video drivers using Microsoft Updates... the update wasn't successful resulting in crashing my Windows 10 with BSOD. There was no way to boot in safe mode, ended up having to reinstall windows 10 from USB and I am still at it reinstalling all the softwares to bring back my PC like it was before.

Even with fresh install of Windows 10 Pro, Cura still doesn't work, so I am giving up for now and is using my other more recent Media Center PC to work with Cura. This PC as a ASUS ROG STRIX Z370-H GAMING Motherboard with Intel Core i5 i5-8400 CPU and powerful video card...

Oh well :o( ...

Asterchades, 2025-06-02:
It's possibly moot at this point, but for posterity I would avoid using the Microsoft facilities to update any of your drivers unless you have no other choice (which pretty much means only Surface owners). I've \*never\* had a completely successful driver installation from this, for any hardware. Often the control panels are missing and something else is broken - and I have also experienced it where the OpenGL HAL was completely absent, though that's likely just a coincidence.

You can disable automatic driver updates through the old *System Properties* control panel. This should be accessible via *About This PC* (search the Start Menu or hit [Win]+Pause), then clicking _Advanced system settings_. From here select the *Hardware* tab, then *Device Installation Settings* and choose the appropriate radio button.

I'm going to close this one for now. This does not lock the conversation, so if you decide you want to revisit this feel free to post back.

closed 2025-06-02 by Asterchades

Asterchades, 2025-06-04:
> I have included 3 screenshots of the NVIDIA Control Panel/Managed 3D Settings/Program Settings to show the options available that I should try to change for the slicer programs.

Just going to touch on these briefly. None of these settings are going to have any meaningful impact in either direction beyond the obvious (eg Forcing MSAA will incur a small performance penalty). Cura, at the very least, doesn't use the GPU for any heavy lifting - it just handles the display. I believe it's much the same for the competition as well but I'm not certain on that. The only setting that might have a noticeable impact on anything is going to be the *Present Method*, but even then that should only be a matter of the the display stuttering or not (DXGI is recommended for Windows 11, but Auto should be fine for most purposes).

I had wondered whether forcing the renderer to one card or the other might have had an effect, but reasoned that the two cards would be using the same driver so there shouldn't have been any potential conflict (unlike a laptop where two separate OpenGL drivers could be in use). It would seem I was possibly right.

> Windows fatal exception: code 0xc000001d

This once again points to an invalid command being issued, either to or by the OpenGL driver. You've pretty much brute-forced every one of the normal solutions by reinstalling your entire OS and all involved software. At this point I have to wonder whether my earlier, seemingly outlandish fixation (which I'd done so well to avoid bringing up) was actually pointing me in the right direction and I should have just followed my instinct.

So, if you'll humour me... first, make sure Windows Update is configured to \*not\* install driver updates automatically. If it is it could immediately undo this. Then install driver 566.36 again. Confirm that 566.36 has been installed through the *NVIDIA Control Panel* using the *System Information* link in the bottom left. Now try Cura again, or any other OpenGL program (including the likes of GLview, which simply polls your OpenGL capabilities). If it still doesn't work (check the Reliability Monitor for more 0xc000001d issues), try driver 552.44. A security update was included with 560 drivers, so I'd feel better using those if they could work, but they may still be problematic.

I have been able to find more than one suggestion that newer NVidia drivers do indeed require SSE4.2 for their OpenGL subsystem, even if it was removed for the DirectX side of things recently. However I cannot find an official source which indicates when this would have occurred (suggestions are either the 570 or 560 branch) so I'm not sure which exact driver to roll back to, and those seem to be the two "pick" versions. This is also third party information as I don't have an appropriate system to test against myself (my Core 2 is long, long gone).

 As an absolute last-ditch effort, you could even try 355.69 (the launch drivers for the GTX 950).

papagino007, 2025-06-12:
Updates...  ***Problems SOLVED***

Well, thanks for your time & support.
Turns out that you were right about the drivers being the problem.
I had tried older driver version 566.36 and it was still not working, but today, I tried 552.44 driver like you had mentioned above (A security update was included with 560 drivers) and VOILA, its working now.

OpenGL now works like it should...

Cheers,

xylite01, 2025-06-25:
@papagino007 Thank you so much for posting the magic version! I've been troubleshooting this for the past couple of days without any luck. I'm using a GTX 1080, also on a Core 2 Quad (Q6600) and Windows 10. It's difficult to find anyone with a similar setup. This CPU is old enough to vote, and I intend to run it at least until it's eligible for presidency.