Ultimaker/Cura #20587: Improvement for details' printing on the 1st layer

yego666, 2025-05-10:
### Is your feature request related to a problem?

When printing the skirt or a brim with a set z-offset usually causes no issues as it is printed with more or less continuous stream of material. Next within such frame some other shapes unconnected with formerly printed frame shape should be printed. A square, a circle or whatever else but no skirt around or inside such shape. In many cases te shape does not stick to bed immediately, leaving a string of floating material. It could be prevented if there was setting letting those feature being printed with slightly lower offset. This would ensure better adhesion for such small features on the first layer.
Perhaps such feature would prove useful.
Thanks for your great work with Cura so far.
Kind regards, Mark

### Describe the solution you'd like

I believe it will solve poor adhesion of small features on the first layer printed with the offset set for brim or skirt.

### Describe alternatives you've considered

Lowering the z distance for the whole layer spoils the whole layer.

### Affected users and/or printers

It will make happy many users like me who are not proffessional "3d printers".

### Additional information & file uploads

I have this problem very often, so the feature proposed.

GregValiant, 2025-05-10:
This is not the first feature request for this.
What you have is really an "Initial Layer Adhesion" problem.  It is often material specific.

The problem with dropping the layer height for the initial layer is that it creates under-extrusion on the second layer.  If your first layer is 0.20 and you drop the height to 0.15 then yes, you will get wider lines and better squish.  The nozzle will move up to 0.40 for the second layer which will mean your second layer is 0.25 thick but the amount of plastic to extrude was calculated for a layer height of 0.20.  By over-extruding the first layer, you under-extrude the second layer.

A good clean bed will often fix the problem but sometimes more help is required.  I don't usually use an "adhesion promoter" for PLA but for some prints with fine features on the initial layer, I find it very helpful to use some.  Adhesion promoter is always necessary for PETG (I print a lot of PETG).

A Cura setting that can help is "Initial Layer Flow".  It will push more plastic and you will get better "initial layer squish".  There is no functional difference between dropping the layer height 25% (like from 0.20 to 0.15) or increasing the Initial Layer Flow 25%.  You will get wider lines and more squish either way.
Another setting that can help is "Initial Layer Bed Temperature".  By starting out with a hotter bed, you might get the plastic to stick better.  If the bed is too hot, you get a pile of goo.
Finally, try turning off the Experimental Setting "Group Outer Walls".  It causes Cura to print all the inner walls first and then go back and adds the outer walls to features.  If you have a lot of small holes to print on the initial layer, the inner walls all get printed and then the nozzle runs all over the place finishing up the holes by printing the outer walls.  Messy.

There are many existing ways to fix the problem.  I've even written a post-processor that does what you are asking (adjust the Z for the initial layer Inner and Outer Walls) but it didn't really help the requesters problem.

I'll go ahead and close this as a "won't do".  I hope you understand.

Here are the Initial Layer Flow settings.  I run mine at 105%.

[image]

closed 2025-05-10 by GregValiant

yego666, 2025-05-10:
As I said, I am no proffessional, therefore I am not aware of many features
and tricks leading to my problem's rectification. Thanks for your
explanation. I'll try all of them right away. You may close the request of
course. Thanks again for all of the Cura. Great job anyway. Have a great
day. Regards, Mark

GregValiant, 2025-05-10:
I would start with "Initial Layer Flow" because it can be adjusted for just walls, or for the walls and bottom skin.  It can help a lot.

Never make a lot of changes all at once.  Better to do one change at a time so you know which one was effective.  You don't want to be chasing your tail.

yego666, 2025-05-10:
Sure thing Greg. Only one variable at a time helps solving an equation.
Besides, how can I chase my tail if it sits in front of me ;) ? I' ll let
you know what will help, and with your leave ask questions in case of
trouble. Thanks again.