Ultimaker/Cura #20588: Unable to print one at a time

barnellio, 2025-05-11:
### Cura Version

5.10.0

### Operating System

Windows 10

### Printer

FLSUN T1 but it is set to a Q5

### Reproduction steps

Follow steps in the link to multiple models but one at a time

https://all3dp.com/2/cura-print-one-at-a-time-explained/

### Actual results

Did not see the "one at a time" or "print before x" options listed in the link

### Expected results

To be able to see the "one at a time" or "print before x" options listed in the link

### Add your .zip and screenshots here ⬇️

[One at a time issue.zip](https://github.com/user-attachments/files/20146639/One.at.a.time.issue.zip)

GregValiant, 2025-05-11:
Thanks for the report.
Set the settings to "Custom".
Click on the visibility tool that is next to the "Search Settings" box.  Set the visibility to "All"

[image]

The settings you want are under "Special Modes".  Set the Print Sequence to "One at a Time". 

[image]

 If you select "Set print sequence manually" just select any model in the "object list" in the lower left of the screen, right-click on it, and then select "print before" or "print after" to change the printing order.

[image]

I'll go ahead and close this.  Let me know if you feel that is premature.

barnellio, 2025-05-11:
I completed the first step but did not see the option "One at a Time" after
searching Special Mode's. The drop down shows nothing
[image: image.png]

What version are you using? Maybe I need to use that one.

GregValiant, 2025-05-12:
Github doesn't allow attachments via email so I didn't get the image you tried to post.
I'm using 5.10.0 but I have several versions installed.  "Print Sequence" has been included in Cura for several years.

The only criteria for Print Sequence is that the printer must only have a single extruder enabled.  If more than one extruder is enabled then "Print Sequence" is hidden.
I don't see any reason why the dropdown box would be empty.  You might have an installation problem.

Go to your configuration folder and find "Cura.log".  Zip that file and post the zip folder here.  You cannot post via email.

GregValiant, 2026-10-01:
I'll go ahead and close this as a setting (or visibility) issue.

closed 2026-10-01 by GregValiant