Ultimaker/Cura #20589: Oops! We encountered an unexpected error during your slicing process.

oldgodzilla, 2025-05-11:
### Cura Version

5.10

### Operating System

Windows 11

### Printer

Anycubic Kobra Max

### Describe your problem and add the package for technical support as a .zip here ⬇️

Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.

Slicing works in 5.6, not in 5.10. I've been having a lot of slicing errors and about 75% of the time I can get them to slice with no problem in 5.6. The rest fail in both. This one in particular failed in 5.10, but worked fine in 5.6

[export-20250511-103411.zip](https://github.com/user-attachments/files/20149707/export-20250511-103411.zip)

GregValiant, 2025-05-11:
Thanks for the report.
I didn't have any trouble slicing the project (BTW it's a about a 29 hour print) so I can't reproduce the problem.
Use the "Help | Show Configuration Folder" command to locate the Cura.log file.  Zip that file and post the zip folder here.  Maybe it will have a clue in it.

[image]

oldgodzilla, 2025-05-13:
Here you go.

[armor.zip](https://github.com/user-attachments/files/20195197/armor.zip)

GregValiant, 2025-05-13:
Thanks for the file.
_"Backend exited abnormally with return code 3221225477!"_
That's a generic error.  The Cura team is aware of that one and it has proven to be hard to find.
I'll mark this as a duplicate report.
UM now gets a separate report when there is a slicing crash so maybe with enough of those they'll be able to chase it down.