Ultimaker/Cura #20590: помилка нарізки

uniko124, 2025-05-11:
### Cura Version

5.10.0

### Operating System

windows 10

### Printer

xviko

### Describe your problem and add the package for technical support as a .zip here ⬇️

однакова помилка у версії 5.7.0 та 5.10.0 "Ой! Під час процесу нарізки сталася неочікувана помилка. Будьте певні, ми автоматично отримали журнали збоїв для аналізу, якщо ви не вимкнули обмін даними у своїх налаштуваннях. Щоб допомогти нам, поділіться інформацією про свій проект у нашій системі відстеження проблем." Модель дуже проста...

[export-20250511-205820.zip](https://github.com/user-attachments/files/20150640/export-20250511-205820.zip)

GregValiant, 2025-05-12:
Thanks for the report.
From the log file:  "Backend exited abnormally with return code 3221225477!"
My understanding is that is a generic crash code.

When I reset the profile back to "Draft" and throw out all your over-rides, the model slices.
If I keep the profile they way you have it set up but change the "Layer Height" to 0.15 it slices.

So it would seem that there is something in the way you have Cura set up that is causing the slice to fail.  I don't know if that is a cause or a symptom.  It shouldn't happen though.

Maybe @HellAholic can find some further information in the log.  I didn't find any errors in the model.

uniko124, 2025-05-12:
Вельми дякую за відповіді!!! Помилка зникла так само як і з'явилася, що трапилось - не розумію. Витратив кілька годин і раптом все запрацювало, як у 5.10.0, так і в інших версіях, хоча зауважу - перед тим як все запрацювало, я видалив версію 5.7.0., а версії 5.6..., та 4.5.0 залишились.

closed 2025-05-12 by uniko124