Ultimaker/Cura #20597: crash

flint-jg, 2025-05-15:
### Cura Version

5.6.0

### Operating System

windows11

### Printer

creality ender 3 v3se

### Reproduction steps

i added a model and hit slice once it fit on  plate
the slicing process crashed and report bug option popped up

### Actual results

i came here to report the bug

### Expected results

i should be able to slice the file

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2025-05-17:
The first step with any slicing problem like this is to make sure you've updated to the latest version of Cura - any fix would require you to update anyway. Every new version in the 5.x branch has addressed slicing faults, with the most common of the known causes having been addressed with 5.7. The current version as of writing is 5.10.0.

GregValiant, 2025-07-03:
@flint-jg Did you follow Asterchades advice and upgrade Cura?  Can this be closed?

closed 2025-07-17 by github-actions[bot]