Ultimaker/Cura #20598: Slicing failed

CAM3RON99, 2025-05-15:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Biqu Bx

### Describe your problem and add the package for technical support as a .zip here ⬇️

I pressed slice and then it gave me an error and said to report the bug. the design is not a complex design

[Gauge Holder Bracket v2 - Copy.zip](https://github.com/user-attachments/files/20224645/Gauge.Holder.Bracket.v2.-.Copy.zip)

GregValiant, 2025-05-15:
Thanks for the report.
There doesn't seem to be any errors in the model, and it sliced fine for me so I'm unable to duplicate the behavior.
This is why we ask for project files and log files.  Please post them.

[image]

CAM3RON99, 2025-05-15:
Hi went back into cura and the message and everything was gone. I resliced
it and it was fine.

What I did do was sliced it first. Made an altercation and the resliced it
and gave me that error message but it has been working fine since then.

Regards
Cameron

GregValiant, 2025-05-15:
Something odd happened.  Since the problem is resolved I'll close this.
If it happens again and if it can be duplicated we can re-open this.

closed 2025-05-15 by GregValiant