Ultimaker/Cura #20606: Slicing calculation error

EmEiBee, 2025-05-18:
### Cura Version

5.9.1

### Operating System

WIN10

### Printer

DIY 

### Reproduction steps

- Open STL from FreeCAD 
- Place and slice
- Control via preview

### Actual results

- 1st try
   In preview all looks well as ever. Transfer the job to the printer result:
- Skirt is partially placed inside the component (see image)
- First horizontal line (back) fits, rest with flow rate reduced to approximately 50% or less in the first layer.

- 2nd try
   Export again from FreeCAD, open with CURA and slice. No options touched.
   In preview all looks well as ever. Tansfer the job to the printer result:
- Skirt is partially placed inside the component again (see image)
- Flow rate seems ok so far

Issue is reproducible

### Expected results

Normal print as ever

### Add your .zip and screenshots here ⬇️

[image]

[Kabelkamm-Gross.zip](https://github.com/user-attachments/files/20274292/Kabelkamm-Gross.zip)

GregValiant, 2025-05-20:
Thanks for the report.
The "flow" issue may be because your build surface isn't flat, or it might not be level.  That can cause thin extrusions but it isn't "under-extrusion" there just isn't enough room for the plastic between the end of the nozzle and the build surface.  If you have a digital caliper or micrometer you can meaure the extrusion.  If it is not your "Initial Layer Height" when you probably have a build surface issue.

Those "I don't know where these come from" lines are from the model.  The base isn't flat and there is in fact a chamfer there.
With the model close to the edge of the build plate you can see what is going on.
[image]

Here I have played with the settings and Cura will put supports in there.
[image]

Here I've painted the angled surface red.
[image]

The skirt will not typically start in the middle of the plate.  The settings are in the Travel section and they are "Layer Start X" and "Layer Start Y".  You can adjust the start point to be where you show it, but it isn't easy.  I think something else is going on with your hot end, or it doesn't like the retraction that Cura sticks in right before the real print starts.

I see extrusions running from the print to the skirt.  That may be because you have the "Skirt Height" at 3, and the "Max Comb Distance with no Retract" at 0.0 which essentially turns retraction off so those lines might well be simple oozing.

Post that Gcode file and I'll look.  I don't see a bug here but rather a combination of:  Your settings, possible problems with the printer, and that angled surface of the model.

EmEiBee, 2025-05-20:
> The "flow" issue may be because your build surface isn't flat, or it might not be level.

I can rule that out. I use Autolevel with 7x7 points, and the nozzle offset is also perfectly adjusted. With completely identical settings, but without the skirt, I get a perfect first layer and a flawless part. When I cross-test with the skirt quite before or quite after, two lines are OK, but the rest again have significantly too low flow. This is reproducible.

>  I think something else is going on with your hot end, or it doesn't like the retraction that Cura sticks in right before the real print starts.

I can completely rule that out, too. As described above, the problem only occurs when I try to print with a skirt; without it, everything works perfectly. This completely rules out any potential problems with the mechanics.
Menawhile I've also printed a few small parts (two plugs for a gear box outlet, a handle for an old lid, and two hooks for a gazebo) that didn't cause any problems, either with, without the skirt nor brim.

> that angled surface of the model.

Yes, that could of course explain the back line. Without the skirt, however, this line isn't printed until later.
I'll remove this unnecessary slant in a V2 version; it doesn't make sense anyway...

I've attached the GCODE from the V1. My settings are already there.

Ty for Assistance

[Kabelkamm-Gross GCODE.zip](https://github.com/user-attachments/files/20348803/Kabelkamm-Gross.GCODE.zip)

GregValiant, 2025-05-20:
Thanks for the files.
I read both gcode files into AutoCad and other than the skirt, they are like peas in a pod.  When I overlay them on each other I can't find any differences.  Geometrically they appear to be identical.  Your second image seems to show extrusions from the model to the skirt, but I don't see that in the Gcode files.

Regarding the flow:  I read the "skirt" file into Excel and looking at layers 1 to 25;  the "mm of filament / mm of extrusion" varies with the line width from .0698mm to .0475mm and the average is .0503mm.  Taking the varying line width into consideration, I don't see any variations in the volume of filament beyond what would be expected.
This is the Cura preview of the "Skirt" gcode file.  This is in "Flow" view.
[image]

This is the same file in the Prusa Gcode Previewer.
[image]

So I don't see anything in the gcode that would be causing any of the problems you are seeing.

Understand that I don't work for UltiMaker.  I am a retired forensic engineer and I just like fooling around with this stuff.  If it's a bug then it's a bug and everyone on the planet would agree that it needs to be stomped on.  I'm just not seeing any evidence of one within your gcode files, or in the gcode file I generated from your project.

I understand that you have our machine dialed in.  I think you should take another really close look at it.  It is extremely unlikely that this is a software problem and that you are the only one who has noticed it.

closed 2025-05-20 by EmEiBee

GregValiant, 2025-05-20:
Yeah, it can be closed.  I'll think on it a bit.  Maybe I'll come up with something else to check.  

To fix that "The skirt doesn't start when it's supposed to" you can use Search and Replace, or this script I wrote (that will soon be included with Cura).  One of its functions is "Adjust starting E location".  It changes that first retraction into a G92 line.  Setting it to "negative retraction distance" will be close to correct.
[image]

[PurgeLinesAndUnload.zip](https://github.com/user-attachments/files/20353894/PurgeLinesAndUnload.zip)