Ultimaker/Cura #20611: Slicing Issue

OniNoDeibbido, 2025-05-20:
### Cura Version

5.10

### Operating System

Windows 11

### Printer

Neptune 3 Max

### Reproduction steps

1. I was slicing and got an error message. 

### Actual results

It gets through about 20% of the process, and says slicing failed.

### Expected results

I slice a file

### Add your .zip and screenshots here ⬇️

[image]

[EN3MAX_Com_Corner_Inner.zip](https://github.com/user-attachments/files/20357584/EN3MAX_Com_Corner_Inner.zip)

GregValiant, 2025-05-21:
There is no model in that 3mf file.
Use the "Help | Export File for Technical Support" command.  Zip that 3mf file and post the zip folder here.  Do not try to send it via email as Github doesn't allow that.

OniNoDeibbido, 2025-05-26:
> file and post the zip folder here. Do not try to send it via email and

EN4MAX_E_Corner_inner.zip
<https://drive.google.com/file/d/1Q7SlV0mApzqygCVjH6tEBoYC9Ph9XBNa/view?usp=drive_web>
 export-20250526-124232.zip
<https://drive.google.com/file/d/1b2tZGaahEPK35-F44vMQ8_KxXkfA_Gd4/view?usp=drive_web>

GregValiant, 2025-05-26:
I don't have access to those them.
Is the project file too big to post here?

OniNoDeibbido, 2025-05-26:
Every time I try to upload the file in GitHub, it gives me an error. I assume it is too big. I tried to upload it through a file share and I get the same result. I have tried other files on Cura, they work fine. I have also reached out to the .stl creators and they are working on their end as well.DaveSent from my iPhone