Ultimaker/Cura #20615: Generating gcode from specific point in preview

CoachB67, 2025-05-22:
### Is your feature request related to a problem?

Hi! :-)

Sometimes a print failes outside of the printers "resume print" parametres. However, one is able to move the nozzle to that specific layer/location in the preview window. A "**_generate gcode from this point_**" would be of tremendous help, both for those of us able to edit the gcode file with fairly good accuracy, but not at least for those of us not that familiar with the intricasies of gcode.

Regards,
Jon Bakken
Sandnes school, Masfjorden, Norway

### Describe the solution you'd like

I believe this will solve the frustrations of having to ditch all the prints having come along far, but have stopped outside the printers "resume print" parametres.

### Describe alternatives you've considered

The alternatives I have concidered is accepting defeat and crying ;-)

### Affected users and/or printers

It will affect everyone experiencing print failures everywhere

GregValiant, 2025-05-22:
There are a few nasty problems that can show up when trying to recover a print.  I wrote an app that will automate it and it usually works quite well - but it still ain't easy and requires human decisions in regards to some recovery options.

1. Where in the Gcode was the failure?  You can measure the height of the print, but if the top is raggedy you will only get an approximation.
2. The stepper motors "disable" and lose their position.  That means the X, Y, and Z must be homed so they can continue.  That often results in a layer shift and/or a "thin or thick" layer because the endstop switches aren't very repeatable.  The "0" this time won't be the same as the "0" was last time.  My printer can be off by up to about 0.4mm in the XY.  That's a fair sized layer shift.  The Z tolerance isn't as bad but there will often be a layer that is too thick or too thin.  "Too thin" is noticeable as a band around the print because that layer over-extruded.  "Too thick" often makes for a weak layer because it's under-extruded.
3. If you weren't there to catch it, the print may have completed, and the bed turned off.  When that happens the print can come loose from the bed (as well as the steppers disabling).  There is no recovery from that failure mode.
4. Homing when there is a print on the bed.  It's a joy.  There must be a place on the bed when the "Z" can home without the print head mashing the print.  It can be done, but usually needs human intervention to make sure.  You must have the print head above the print and then you can home the X and Y at that height, travel to the open spot, and home the Z.

Continuing a print is like any Gcode "file splicing" except you are basically throwing out the first part of the print.  Splitting the file and throwing out the layers you don't need, then possibly home and use G0 moves to get up above the print, move to the "XY start" and use G92 to set the "E" location and possibly the "Z" location, and then just run the file.
It's more tedious than difficult and it's a skill that needs to be practiced.  _If you take to long to re-start the print, it's usually not salvageable._

This was just a test model.  The layer height changes from 0.20 to 0.10 to 0.30.  The difference between this spliced Gcode and a spliced recovery gcode is that it wasn't necessary to re-home at any point.  That's why the transitions at the layer height changes are exactly right.
[image]

This is a valid feature request.  I just don't think it's something you will see anytime soon.

HellAholic, 2025-05-26:
Here's a rundown of why this is not a feasible feature to implement:

1. Generating the gcode is possible but homing and bed leveling is not. In addition the nozzle needs to prime correctly and wipe on the build plate (skirt / brim /purge line /etc ).
2. The generated gcode after each slice is not fully deterministic, meaning each time you slice, some minor differences will be there in how a certain layer is printed
3. High risk of failure if any of the measurements are wrong which leads to more time spent and more filament used without a good result
4. Risk of damage to the printer due to an unsafe process, if this was a feasible feature, manufacturers would have placed it as a Firmware feature to continue print from layer on the same gcode.

GregValiant, 2025-05-26:
Recovering a print isn't impossible, but it does require a lot of forethought.

This is the main recovery dialog in my printing app.  (The app communicates with the printer through the USB but tells the printer which SD card file to print so it isn't printing over the USB, just receiving and echoing Gcode commands I might send.
The Gcode file on the hard drive must be identical to the one on the SD card or the "Byte location" won't match.
[image]

Once that gets figured out, then the "resume-from" numbers get transferred to the main Print From SD Card dialog.
[image]

Then I generate the recovery script (that will run prior to resuming the original gcode file) with this dialog box.
[image]

The script shows up here for the user to check.  In this script I decided I didn't need to home the axes.  That can happen when I luck out and am close to the printer when the failure starts.
[image]

Finally I hit the "Print from Byte Location" button and cross my fingers.  The recovery script gets sent to the printer over the USB and the printer sees the "byte location" to resume from and starts at that location on the SD card file and off it goes.
My success rate is near 100% _UNLESS_ homing is required.  Homing will lower the chances of a good recovery to about 15%.

Not for the faint of heart, but I've saved a couple of long prints that would have been junk without the ability to generate the re-start script.  My firmware requires the DOS 8.3 filename.  That by itself can be a show stopper for most people.
These two lines were not easy to come up with.
M23 \ALL_SK~2.GCO    ;DOS 8.3 FileName
M26 S779798               ;Byte location to resume from

HellAHolic has marked this request is a "won't do" so I'll go ahead and close it.  I hope you understand.

closed 2025-05-26 by GregValiant