Ultimaker/Cura #20616: Slicing bug

jujumo, 2025-05-22:
### Cura Version

5.9

### Operating System

Win11

### Printer

ender3

### Describe your problem and add the package for technical support as a .zip here ⬇️

[tarot - card-box-cap-1.zip](https://github.com/user-attachments/files/20395947/tarot.-.card-box-cap-1.zip)
This files generates a slicing bug when layed flat. Slice works on other orientations.

GregValiant, 2025-05-22:
Thanks for the report.
I tried a couple of things but I can't get it to NOT slice.  Often, when the orientation results in different results, there is an error in the model.  In this case the model appears to be error free.
Use the "Help | Export File for Technical Support" command.  Zip the resulting 3mf file and post that zip folder here.  It will contain a Cura project file and the cura.log file.

jujumo, 2025-05-23:
OK, thanks for looking into it.
Here is the export along with stl and a screenshot.
I change the stl model in a way it is produces the error right away (no need to rotate).

[image]

The weird thing is that the model works if I import it from the exported 3mf.

I hope it helps.

[v8.zip](https://github.com/user-attachments/files/20405879/v8.zip)

GregValiant, 2025-05-23:
This is a generic slicing error from your log file:
_Backend exited abnormally with return code 3221225477!_
I've seen that one a lot in other bug reports but I've never been able to duplicate it on my system.
The Cura team will take a look.