Ultimaker/Cura #20617: Cura 5.10.0 Crashing on Windows 11

stone178, 2025-05-22:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Ultimaker S7

### Reproduction steps

1. uninstalled Cura 5.4.0
2. installed Cura 5.10.0
3. Open Cura
4. open recent file made with Cura 5.4.0
5. click on "open project"

### Actual results

after clicking on "open project" or "import model" the software abruptly closes with no warning or messages. 

### Expected results

The model should open so I can work on it.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/20399172/cura.log)

GregValiant, 2025-05-22:
Thanks for the report.
I'm not very good at the log files but there are a lot of QML errors.  @HellAholic will take a look.

I'll take a guess that something didn't get installed correctly.  We'll see how good I guess.

stone178, 2025-05-23:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> ---- External Email: Use caution with attachments, links, or sharing data ----

Great, thank you! I appreciate your help with this

Sincerely,

Amanda Stone
Mechanical Engineering PhD Student | Herrick Laboratories
Graduate Relations Chair | Purdue SWE
Head Graduate Advisor | IN-A Tau Beta Pi Honors Society

stone178, 2025-05-27:
Hi @HellAholic, thank you for the suggestions above!

I went to step 8 and found multiple crash logs for Cura. I just reinstalled it and this is the crash log recorded after trying to open a file:

Faulting application name: CuraEngine.exe, version: 1.0.0.0, time stamp: 0x67dd5a29
Faulting module name: ucrtbase.dll, version: 10.0.22621.3593, time stamp: 0x10c46e71
Exception code: 0xc0000409
Fault offset: 0x000000000007f6fe
Faulting process id: 0x0x10CD8
Faulting application start time: 0x0x1DBCF2C68AF15A1
Faulting application path: C:\Program Files\UltiMaker Cura 5.10.0\CuraEngine.exe
Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
Report Id: 2dcfcd37-d9db-49db-97c1-6b949dd383bc
Faulting package full name: 
Faulting package-relative application ID: 

Any suggestions on how to proceed would be great. Thank you again for your help!

stone178, 2025-05-27:
Hi @HellAholic,

Thank you for the response. I double checked opening a benchy stl file and I had no issues with the software crashing. I also discovered that if I open the file and click on "import models" then I have no issues, but if I click on "open project" then it crashes. The model I am working on is a composite with multiple models, so this means that I cannot save my configurations of the various models if I just import them.

I tried using various printers as suggested in #2 and I get the same issue.

For #3, I noticed I have app data files for 1.2, 5.4, and 5.10 even though I only have 5.10 installed currently (I uninstalled the others). Could this be leading to any issues?

I' tried attaching the .3mf file for reference, but it looks like that file type is not supported, so if there is another way I can send it to you let me know.

Thank you for your help figuring this out!!

stone178, 2025-05-27:
Got it, here is a compressed folder with the file. Let me know if you need anything else. My hope was to create print settings that I can continue to iterate and improve using a file like this. Thank you!

[UM7 test.zip](https://github.com/user-attachments/files/20466408/UM7.test.zip)

stone178, 2025-05-28:
Thank you for these suggestions. I was able to delete the older versions but cannot delete the 5.10 folder because this was saved on a network location drive (which might be the cause of my problems) and I don't have permission to delete it. I am working with my institution's IT dept to resolve this. 

On the last time I ran Cura and it crashed, I found the following text file in the AppData location that I believe lists the errors. I am sharing that here in case this helps you at all.

Thank you again for all of your help, I really appreciate you assisting me with this over the past few days.

[stderr - Copy.zip](https://github.com/user-attachments/files/20495798/stderr.-.Copy.zip)

stone178, 2025-05-29:
Thank you again for all of your help, I really appreciate it! I'm still awaiting assistance from IT, but I found that the 5.10, etc files are in both my network and local drives, so I'm wondering if that might also be part of the issue. I have no idea why they were saved to both places but I will let you know what I find out from IT.

stone178, 2025-06-04:
Hi @HellAholic, I have been working with the IT department and they have identified that the issue is likely that the local app data file is being saved to the network drive rather than the local drive. They said that there is nothing they can do to fix it on their end, so I wanted to ask if there is a setting or some way for me to set the drive where Cura saves this data.

Thank you!

HellAholic, 2025-06-04:
Hey @stone178 ,
Cura uses `os.getenv("LOCALAPPDATA")` and `os.getenv("APPDATA")` for the locations of the cache and configurations. So unfortunately I'm not familiar with any way or solution to hard-code the path in Cura to modify the location. I tried a couple of tricks on my local machines (editing the registry, system environment variables, etc) but nothing seems to work.

There is this one trick that might just work but it's a very slight might if you haven't tried it before.
1. If you have Cura installed, Remove it from your system, and Clear out the non-network APPDATA/cura and LOCALAPPDATA/cura folders.
2. Hopefully you still have the installer on your device, if not download the latest version you want to use. 
3. Restart your computer (optional - but at this point why not :P)
4. Disable Wifi, disconnect from all networks (Ethernet - wifi - ...) so you no longer have access to internet or the Network Drive in question
5. Install Cura
6. Open Cura
7. See if the configuration and cache are created locally in the appdata and local appdata -> test with your project file and see if you still get a crash.
8. Restore network connection, close Cura and open it again -> retest to see if you can open the project file still.

stone178, 2025-06-04:
@HellAholic Thanks for the suggestion, I made it to step 5, but when I tried to install Cura without ethernet or network, I got the following error about not being able to find the network (U) drive:

Traceback (most recent call last):
  File "cura_app.py", line 96, in <module>
  File "<frozen os>", line 215, in makedirs
  File "<frozen os>", line 215, in makedirs
  File "<frozen os>", line 215, in makedirs
  [Previous line repeated 1 more time]
  File "<frozen os>", line 225, in makedirs
FileNotFoundError: [WinError 53] The network path was not found: 'U:\\'

I appreciate any suggestions or insights you may have!

HellAholic, 2025-06-04:
Did you use the installer.exe or the installer.msi?

In case of the .msi, select Custom installation and choose a local install destination

[image]

[image]

Browse -> Select a local path that is not on U:\...
Example that I have on my system is `C:\Program Files\UltiMaker Cura 5.10.1`
It could also be `<local_drive>:\UltiMaker Cura 5.10.1` or any path you prefer

The .exe installer also include the install path but it's part of the normal flow so you will see it either way:

[image]

Same thing here -> choose a path that is not on U:\...

stone178, 2025-06-04:
@HellAholic I used the exe file, and I selected the destination folder to be on the local (C) drive, and I still got the error message above when I tried to install Cura without ethernet