Ultimaker/Cura #20622: me again

familyplans69, 2025-05-24:
### Cura Version

5;10

### Operating System

windows 11

### Printer

anycubic kobra 2 neo 

### Describe your problem and add the package for technical support as a .zip here ⬇️

it will slice my object in one orientation but refuses to slice it when i put it flat on the bed (which is the way i want it to , sorry but i have no idea how to send logs or anything

GregValiant, 2025-05-25:
I'll close the other report.
When a model won't slice in a specific orientation it is often because there are errors in the model.  You need to check the model for errors in an app like MS 3D Builder.  There are also on-line repair sites.

Load the model in Cura and set up the slice.  After it fails - use the "Help | Export file for technical support" command and then "Open Folder".
Right click on the new 3mf file and select "Send to..." and then "Compressed (zip) folder".
Post that zip file directly here (do not try to email it).

familyplans69, 2025-05-25:
i think i have done it , and i love cura and this must be the first time i
have had a problem , keep up the good work

GregValiant, 2025-05-25:
Please read my previous comment.  Take special note of where it says "Do not try to email it".  Github rules are that files must be directly posted.

familyplans69, 2025-05-25:
[export-20250525-085451.zip](https://github.com/user-attachments/files/20431380/export-20250525-085451.zip) eek i didnt even realise it was on here !!