Ultimaker/Cura #20627: Cura treating void as solid

BuschyX, 2025-05-25:
### Cura Version

5.10.0 and 5.8.1

### Operating System

Win 11

### Printer

Aquila C2 (modified)

### Reproduction steps

Load .Stl
Slice/Preview .Stl

Cura seems to be only Slicer having issues with this file.
I've tested it in multiple Slicers and they Slice as intended

### Actual results

Preview/Slice shows parts of model as solid when it's not.
Prepare shows Model as intended.

### Expected results

A Sliced Model that is not solid and has supports as intended.

### Add your .zip and screenshots here ⬇️

[New-Tool.zip](https://github.com/user-attachments/files/20432602/New-Tool.zip)

BuschyX, 2025-05-26:
Turning off "make overhang printable" seems to have solved the issue.

GregValiant, 2025-05-26:
Thanks for the report anyway.  I'll go ahead and close it.

Both "Make Overhangs Printable" and "Remove All Holes" are specialty settings that are good when you need them, and bad when you forgot you turned them on.

closed 2025-05-26 by GregValiant