Ultimaker/Cura #20644: "Gradual Infill Steps" does not actually increase infill density

xDSCx, 2025-05-29:
### Cura Version

5.10.0

### Operating System

Windows 10

### Printer

Anet A8

### Reproduction steps

I tried using gradual infill steps to reduce material usage in larger prints.
After slicing I found that instead of adding additional infill lines, the added lines replace the old ones. 
I tried that with almost every infill pattern and always got the same result.

### Actual results

In the picture you can see, that the infill lines before the step are no longer printed after the step, which means the infill density is not actually increased:

[image]

### Expected results

the additional infill lines AND the old infill lines are printed, doubling the density

### Add your .zip and screenshots here ⬇️

[Profile.zip](https://github.com/user-attachments/files/20506638/Profile.zip)

GregValiant, 2025-05-29:
Thanks for the report.
I can't duplicate this.
This is with "Connect Infill Lines" turned on and with 10 "Gradual Infill Steps".

I only checked with Grid and Gyroid and didn't see the problem.  The density increased with each step.
This is one layer at a time.
https://github.com/user-attachments/assets/2ee6e072-cfac-4da6-8447-c09184ae9f4f

HellAholic, 2025-05-29:
It is possible to replicate if you have a model with multiple top surfaces, I raised it previously (it's also the same behavior for the top surface pattern).
I have to dig through the backlog to see what was the fate of the ticket.

[image]

[image]

[image]