Ultimaker/Cura #20647: Orientation failures

Chris3126, 2025-05-31:
### Cura Version

5.10

### Operating System

Windows 10

### Printer

Anet A8

### Reproduction steps

In Cura 5.10 the automatic orientation and the options under extensions don't use the most logic and obvious orientation. 

[image]
[image]

### Actual results

Every time one must rotate the object.

### Expected results

After rotation the parts manually they print perfect.

### Add your .zip and screenshots here ⬇️

See above. In all cases the manual rotated parts print perfect.

HellAholic, 2025-05-31:
If you want automatic orientation, download the plugin Auto-Orientation either from the Marketplace in Cura or the github page https://github.com/nallath/CuraOrientationPlugin

Then you can set Automatic Orientation from Extensions > Auto Orientation > Modify Settings

[image]

[image]

[image]

closed 2025-05-31 by HellAholic

Chris3126, 2025-06-01:
The auto orientation plugin was installed!! The part was wrongly placed even without using the extension. I have deleted the extension and now the standard Cura places the part correct.

HellAholic, 2025-06-01:
Glad that resolved the issue one way or the other (adding/removing), if you have a good example models of what the orientation should be and how the plugin places them you could submit a issue for the plugin in its repository here https://github.com/nallath/CuraOrientationPlugin