Ultimaker/Cura #20658: print layers and sides all outta wack

Timby2009, 2025-06-05:
### Cura Version

5.10.0

### Operating System

Mac OS Ventura

### Printer

Elegoo Neptune 3 pro

### Reproduction steps

I started by downloading the  STL files, "Picnic Table" by "LeoQuindos" from thingiverse, and then sliced them with 15% infill and two line wall thickness, the nozzle size is 0.4mm and the layer height is 0.2mm,the bed adhesion was set to brim and all the other setting were factory, except the z seam was randomized.
I then sent it to my printer and let it run for about 55 minutes while I had supper. when I came back the sides and bed adhesion that were supposed to be straight were all bowed out but the rest of the print looked normal. I stopped the print and put the g-code back into cura and the result that i got was surprising, the bed adhesion was a complete mess and didn't resemble what the bed adhesion should have looked like at all. The layer alignment were all off and the infill was all piled up. it was a very different result then I was getting with the print. I thought that it might have just been a one off glitch and reimported and sliced the same files, in the preview it looked fine, but once i exported the G-code and then re imported (if that's the right word) the G-code, the result was worse than before. 
I then downloaded the "XYZ 20mm Calibration cube" by "iDig3Dprinting" and placed that i cura. after turning off the supports, I sliced the Stl file and exported the G-code. I then imported the G-code and checked the preview, the cube resembled a cube, but most of the details are missing, all the infill is out of whack and slicer is missing parts of the walls. the top three layers are also floating quite a ways off the rest of the cube.
I then reset all the cura setting to factory and resliced the file. the preview was expected but when I added the G-code back into cura, the print looked fine.

### Actual results

explained above.

### Expected results

explained above.

### Add your .zip and screenshots here ⬇️

[image]
[image]
i didn't save any of the slices of the XYZ cube. I also didn't save the G-code of the first attempt print. Sorry if that makes it harder to thy to fix the problem.

Timby2009, 2025-06-05:
thank you everyone for helping me find the problem. I sliced the files again and when i put the G-code back into cura it looked fine. I am going to print the g-code and I will comment when it is done. thanks again.

closed 2025-06-05 by Timby2009

GregValiant, 2025-06-05:
Perfect.
A word to the wise - when you start a new project you should "re-select" the profile you are using so you can check what settings you may have changed.  If you "discard" changes you will start from the base of whatever profile you are using.
Settings like "Arc Welder", "Remove all Holes" and "Make Overhangs Printable" have their uses, but they aren't settings you want turned on all the time.  It's easy to forget.