Ultimaker/Cura #20661: Slicing failed

Cracoukas, 2025-06-05:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Voron V2.4 350 with klipper

### Reproduction steps

1. Slice object

### Actual results

Slicing error

### Expected results

Display sliced model

### Add your .zip and screenshots here ⬇️

[V350_Volant - Impact PLA - REVO HF 0.4.zip](https://github.com/user-attachments/files/20618314/V350_Volant.-.Impact.PLA.-.REVO.HF.0.4.zip)

[cura.log](https://github.com/user-attachments/files/20618338/cura.log)

GregValiant, 2025-06-06:
Thanks for the report.
This is from the log file.
_"Backend exited abnormally with return code 3221225477!"_
That's a generic code and could be several things.  It's definitely a crash.
There doesn't appear to be anything wrong with the model.

When I switch to the "Fast" profile the model slices.  I cannot get it to slice with the "Fiberology Impact PLA" profile.  There is something in there that Cura doesn't like.

This is "Fast" with supports turned on.
[image]

The Cura team will take a look.

HellAholic, 2025-06-06:
Seems like an issue in the tree support calculation/generation. I'll add this model as a test case for the tree support fixes.
If you move the model slightly from (X=0, Y=0, Z=0) to (X=1, Y=1, Z=0) it should slice with your set profile.

[image]

Cracoukas, 2025-06-07:
Thanks, it works !

Sincerely

Jacques Maeder

Route de la Sauge 6J / CH – 2019 Chambrelien

Tél. +41 (0) 76 530 15 01

eMail : ***@***.******@***.***>

______________________________________________________________
Ce courriel peut contenir des informations confidentielles. Il est destiné uniquement à la/les personne/s auquel il est adressé.
Toute distribution, retransmission ou autre utilisation pour des personnes autres que le destinataire originel est interdite.
Si vous recevez ce courriel par erreur veuillez s'il vous plait en aviser l'expéditeur et détruire toute copie.

image.png (view on web)<https://github.com/user-attachments/assets/4b410f86-9fa9-45aa-9620-fa3af6cfe4a9>