Ultimaker/Cura #20662: Market place Red Dot

gin8330, 2025-06-06:
### Cura Version

5.10.0

### Operating System

Windows 10 pro

### Printer

Artillery X2

### Reproduction steps

I did nothing ,because it needs ad code .

### Actual results

If it is the same changes in "plugins" or "materials" that irritating dot appears. Irritating is because you ,as a user, are in a helpless situation.
You can scroll up or down in  "plugins" or "materials" and have no indication what this "red dot" marks.

[image]
[image]

### Expected results

I hope you will make some changes as soon as possible, because this bug show as  "Cura slicer" looks like unfinished product.
Actually, this "red dot" travels through Cura slicer version. I'm surprised, as no one has issues with it.  

### Add your .zip and screenshots here ⬇️

Thank, you

GregValiant, 2025-06-06:
I usually get something like this.  There is an update button.  I recall having to look for which one has been updated.  In this case it came right up.  Myabe a "Has been updated" filter could be applied.

[image]

There is a setting here that you could turn off:

[image]

I don't know that either one would help.  You should only get the updates for plugins you have installed.

gin8330, 2025-06-07:
Thank you, guys, for the explanation. I knew all about red dot when getting updates ,update button beside plugins. It will be ok if that worked . My personal situation is:  one computer at home, another at work in both installed last version of Cura Slicer. Both Curas are synchronized through account at  UltiMakers server. The Odd thing is Cura slicer at work has red dot and no any indication for plugin updates ,Cura slicer at home-no red dot.

fieldOfView, 2025-06-07:
In the list of installed packages, try scrolling down to the installed _materials_. The dialog lists plugins and materials separately, and if there is a material profile waiting to be updated, it will be buried by all plugins that ship with Cura.

gin8330, 2025-06-09:
@fieldOfView  There is the point . With in finished software ,you should have to scroll any ware ....I should pop up medially what part plug ins or material advised to update or it is same changes.

gin8330, 2025-06-09:
@fieldOfView  Thank you,it was a good suggestion.  😍