Ultimaker/Cura #20668: Infill sometimes prints after wall when using modifier (and sometimes before as normal)

earth75, 2025-06-07:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Reproduction steps

1. Put several meshes on the bed
2. Add a modifier cube and set ti so that it overrides the infill at its intersection with the normal meshes example below
[image]
3. make sure it 

### Actual results

3. Sometimes the infill will be added after the perimeters like below
[image]
 And sometimes not (note the presence or abscence of the green perimeters in the preview)
[image]

Its unclear why it would behave differently for different meshes on the same plate.

### Expected results

Given the same settings, the infill wall order should always be the same and should not change from model to model.

### Add your .zip and screenshots here ⬇️

[AK_ruche V4.zip](https://github.com/user-attachments/files/20640251/AK_ruche.V4.zip)

[image]

[image]

[image]

GregValiant, 2025-06-09:
Thanks for the report.
I agree there is a bug in there somewhere.  I tried a few things hoping to at least come up with a workaround.  No luck.

Two models always print "Walls First".  I have no idea how or why something like that would occur.  I tried deleting the models and replacing them with new, I tried moving the left models to the right and the right models to the left.  I rotated the modifier and it's 4 models 180°.  Still no luck.  Two of the models always print "Walls First".

@HellAholic - This is a different sort of thing.

HellAholic, 2025-06-09:
There is an issue with the infill only mesh when it involves more than 1 model-mesh.

Workaround:
You can use a cutting mesh instead of the infill mesh, set the top layers to 0 for the cutting mesh.

[image]