Ultimaker/Cura #20672: Slicing fails repeatedly

marmotzero, 2025-06-09:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 Pro

### Reproduction steps

I'm trying to slice supported files from the Wilderwood Circle from the Printable Scenery Gloaming Swamp series: https://www.printablescenery.com/product-category/previous-campaigns/the-gloaming-swamps/. Specifically, I'm printing the supported version of the leaf clusters for Tree A. I import the STLs, arrange them on the plate, load my printing profile, and press slice. 

### Actual results

I get an error message: "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

### Expected results

Successful slicing which I could then export to a thumb drive and take to my printer

### Add your .zip and screenshots here ⬇️

[Cura.log.zip](https://github.com/user-attachments/files/20660463/Cura.log.zip)

GregValiant, 2025-06-10:
Thanks for the report.
Both Cura 5.5 and 5.6 had real problems slicing both "organically shaped" models, and models with rough hewn wood or stone floors.
This is the UltiMaker Cura site.  If you are using "Elegoo Cura" then you need to make a report on their site.  
I'll mark this as a duplicate and go ahead and close it as those slicing errors were a well known issue.

closed 2025-06-10 by GregValiant

marmotzero, 2025-06-11:
Since Elegoo Cura hasn't upgraded past 5.6.0, sounds like I need to switch to Ultimaker!