Ultimaker/Cura #20673: Slicing Error - Windows 11, has persisted since v5.5

musicartist525, 2025-06-09:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

FLSun V400

### Describe your problem and add the package for technical support as a .zip here ⬇️

Hello,

I receive the "Slicing Failed" error on nearly every file since v5.5.  I can only use v5.4, and this is on two computers, both running Windows 11.  All profiles that I've tried result in the error.

Please let me know if I should send more packages or if there is a workaround.  Thanks!

[export-20250609-181324.zip](https://github.com/user-attachments/files/20661932/export-20250609-181324.zip)

GregValiant, 2025-06-10:
Thanks for the report.
The log file is showing a lot of errors.  
@HellAholic  from the Cura team will need to look at this one.

HellAholic, 2025-06-10:
This seems like a corrupted configuration. Unclear what has caused it but if you see a message from Cura with a ❌ then it indicates there is an error and you should not ignore it.

Simple fix that will remove all printers, profiles, plugins, etc:
- Remove the configuration folders and configuration cache folders in `%appdata%/cura` and `%localappdata%/cura` (e.g 5.3, 5.4,...,5.10)

More complex fix:
- Remove all the folders except 5.4 (last working version)
- Isolate the plugins -> you can reinstall them later so removing them makes things easier.
- Examine the contents to see what file could be the cause