Ultimaker/Cura #20678: program crash/close while printing over usb during screen lock

goofyseeker311, 2025-06-11:
### Cura Version

5.10.1

### Operating System

win 11

### Printer

elegoo neptune 3 pro

### Reproduction steps

1. print a model for 3 hours, 2 hours left.
2. lock windows with winkey+L

### Actual results

3d printer stopped printing shortly after lock screen activation.
after unlocking, cura program was no longer open.

### Expected results

lock screen should not affect 3d printing over usb in any way.
cura program should not close itself during lock screen printing.

### Add your .zip and screenshots here ⬇️

the model is such that it takes total of 5-6 hours to print. model itself is not relevant.

GregValiant, 2025-06-11:
Thanks for the report.
"USB Printing" is a legacy feature that is kept in Cura because the "Ultimaker Original" printers require it.  My understanding is that UM abandoned USB printing it because it was very glitchy.  It has not been updated in a long time.  If a developer was to fix it, they would be an instant hero.  That hasn't happened.  There are too many different hardware combinations, firmware combinations, and driver combinations, to get it to work reliably.

If you want/need remote printing you would need to move to Octoprint or install hardware to give you network capability.
I'll mark this as a "won't do" and go ahead and close it.  I hope you understand.

closed 2025-06-11 by GregValiant

goofyseeker311, 2025-06-11:
the program should not close at all if the windows is in the lock screen, no matter if usb printing or any other, tho.

goofyseeker311, 2025-06-11:
I'm just chatting about the usb support, and true my elegoo neptune 3 pro works over usb just fine. its a current 3d printer. I thought cura was for all the printers, not just ultimaker printers. has the scope narrowed to just ultimaker printers after cura transferred to ultimater ownership? I like to end this discussion now. lets not continue.

goofyseeker311, 2025-06-11:
I only get an implied feeling of the purpose of cura. not really reading any small print.

ie "State-of-the-art slicer app to prepare your 3D models for your 3D printer.
With hundreds of settings & community-managed print profiles,
Ultimaker Cura is sure to lead your next project to a success."

implies what I stated above. I'm not responsible for your assumed goal, ie any implementations.
just asking if your goal is what you state it is. the general feeling implication of the main goal.

all other applications, ports etc dont get affected by screen lock. as expected. so cura behaviour was unexpected.
usb thumb drives etc stay attached to the user that was using them, and stay running on the background.

but the original issue, cura closed itself completely and as a result stopped all print actions, because it was closed.