Ultimaker/Cura #20688: [5.10.0] Randomly starts in Spiralize Outer Contour

FucinaLab, 2025-06-13:
### Cura Version

5.10.0

### Operating System

Windows 10

### Printer

Creality CR10S Pro V2

### Reproduction steps

1) Slicing an open surface with Surface Mode and Spiralize Outer Contour

### Actual results

The result is almost perfect except that unexpected starting points appear randomly

[image]

### Expected results

No random starting points should appear

### Add your .zip and screenshots here ⬇️

[CCR10SPRO_GEOWORLD 05.zip](https://github.com/user-attachments/files/20733300/CCR10SPRO_GEOWORLD.05.zip)

FucinaLab, 2025-06-14:
I can confirm that the file can be printed ( see the attached image of another model that I slice in the same way ). 

[image]

However, in some places the path is not continuous, while it should be unique.

FucinaLab, 2025-06-14:
Here are more pictures of a printed vase with the same settings

[image]

[image]

GregValiant, 2025-06-14:
What did you slice that in?
This is from the image above.  It sure looks like a Z-seam that would be produced with "Surface" mode when "Spiralize" failed to work.
[image]
This is from 5.10 and looks a lot like your print.
[image]

There is no "warning" when "Spiralize" fails.  You just get starts-and-stops and sometimes solid extrusion moves from one area to another.
When you have built-in islands (like that model does with it's concentric surfaces), there is no way for the nozzle to move from one "island" to the next "island" without stopping the extrusion, and restarting.  By definition, that is a failure of spiralize.  The fact that the islands are very close together does not change that.

This is your project with no changes, in 5.10.0
There are three concentric walls, and a Z-seam.  Spiralize failed but "Surface" was turned on so it sliced.
Here you can see that the nozzle is moving from one surface to another without stopping.  That is wrong and is a product of the model design and the Cura settings.
[image]

When all three "train tracks" are taken into account, you can see that there is this cross-over spot.  I would guess that in certain cases this would be where a "start" was located.
[image]

This model can be spiralized.  I have doubts as to whether it can be _consistently_ spiralized.  A typical model for Vase Mode would be a single solid model and not a collection of concentric surfaces that have no volume or thickness.  You may occasionally get what you expect, but then again you may not.

I would go this way with the model design.
[image]

Which results in this fully spiralized model with the flange on the base.

[image]

The bug label remains on this (although I don't see one myself) and at some point in time it may be addressed.  Your best bet is to figure out a workaround which might entail using a different slicer, or changing the method you use to design these sorts of models.

FucinaLab, 2025-06-15:
I'll take a step back and explain what my goal is: spiralize an object being able to choose the thickness while maintaining the standard nozzle diameter of 0.4 mm.
I'll also describe the process I used to create the 3D object:
1) Closed curve from a 2D graphics software saved in a .SVG file.

[image]

2) Simplification of the curve, offset with a multiple distance compared to the nozzle diameter and creation of a closed self-intersecting curve with Grasshopper (Rhinoceros).
3) Loft of the curve in the Z direction to create the walls of the vase + loft with a larger offset to create the base of the vase.

[image]

4) Export in .STL format of the two open and self-intersecting surfaces.

[image]

5) Slicing with Cura using the Surface Mode and Spiralize Outer Contour. 

[image]

This way I can create a continuous path that allows me to speed up printing with a wall thickness of my choice.
I don't understand. If the procedure is wrong or the .STL file has problems, the slice should fail. Instead, the slice works, but the result is not consistent with the initial settings. Why does Cura break the path at random points along the Z-axis? The detail you showed me is not the problem; I set the base curve this way. For me, this is not a bug, but some sort of method signature. The problem, in the project I shared, is what happens at levels 2, 3, 4, 216, 217, 393 and 394.
It seems that the path cannot be longer than a value, but that doesn't make sense.
I'm surprised that no one has ever needed a tool in Cura that can spiralize objects by choosing their thickness. The only safe way to do this is to create a G-code file without using standard slicing software, but I wanted to avoid that option for now.

I started using the Surface Mode feature by taking inspiration from a file shared on Thingiverse ( https://www.thingiverse.com/thing:5043958 ) since, for a workshop aimed at children aged between 7 and 10, I needed to quickly convert 2D drawings into a printable 3D file

[image]

Now I would like to be able to control the thickness...

GregValiant, 2025-06-15:
Take a piece of paper and a pencil.  Put the point of the pencil on the paper and draw a shape.
- You are not allowed to lift the pencil from the paper.
- You are not allowed to cross a line.
- You are not allowed to backtrack on a previously drawn line.
- Using those rules eventually bring the pencil back to the start point.

That image of the leaf violates the rules and is clearly not suitable for spiralize.  When a line crosses itself it creates a loop and a loop is a single lap around an area.  When the nozzle is done with the loop what would happen?  It must backtrack (illegal) or lift and travel (also illegal).

I've seen some model airplane wings and fuselages that are really well done.  They have internal ribs and spiralize perfectly.  If you can find one of those models it will give you a boost on designing your own.  Ribs must be designed so that there is a "to and from" extrusion possible.  Consider times on a comb.  They can be spiralized if each tine is wide enough for two nozzle passes.

I generally spiralize with a 0.4 nozzle and a 0.6 line width but I have pushed it to a 0.8 line width.  That worked well.  I could go up to a 1.0 line width on the 0.4 because that is the size of the flat at the bottom of the nozzle, but so much plastic through a small hole is tough to do.  You should be able to print up to 0.70 line width without having to slow down too much.  Spiralize often causes very short layer times and so the Minimum Layer Time will often come into play.  It isn't usually a very fast process.

Your method of designing looks alright until the multiple surfaces show up.  Cura must "lift the pencil" to move from surface to surface because there is no connection between them.  In effect, each surface is an island.

Here is the model I made up.  This one will spiralize as it is comprised of a single piece.  At 0.7 line width I see that the "top layer of spiralize bug" shows up.  That is a known bug and will be fixed in 5.11.  I just delete the top layer from the gcode.

[OGEE Vase.zip](https://github.com/user-attachments/files/20746398/OGEE.Vase.zip)

FucinaLab, 2025-06-15:
In summary, the "condicio sine qua non" for using Cura's Spiralize command is to have a continuous, non-self-intersecting surface.
If these prerequisites are not present, the result cannot be expected to be reliable.
But Cura, somehow, spiralizes "something" in accordance with this maxim: "According to all known laws of aviation, there is no way that a bee should be able to fly. Its wings are too small to get its fat little body off the ground. The bee, of course, flies anyway. Because bees don't care what humans think is impossible."

GregValiant, 2025-06-15:
Bees don't actually "fly" in the sense that birds do.  Bees have small naturally occurring organic anti-gravity devices in their thoraxes and so they use their wings as fins and "swim" through the air.

FucinaLab, 2025-06-17:
If surfaces is a niche thing that always causes problems, perhaps it should be eliminated, as was done with Wire Printing. Otherwise, as a user of Cura, I should be able to use this feature and report that it does not work as well as in previous versions.
However, this simplification will lead you toward software like Bambu Studio, which is perfect for the average user but does not allow the same level of customization that Cura offers.

closed 2025-06-21 by FucinaLab