Ultimaker/Cura #20692: [5.10.1] Cura asks me to go through the initial setup every time I launch it.

wdiasjunior, 2025-06-15:
### Cura Version

5.x.x

### Operating System

Arch Linux

### Printer

Ender 3 V2

### Reproduction steps

1. Open Cura
2. Setup an account and printer
3. Close Cura
4. Open Cura

### Actual results

Cura asks me to go through the initial setup every time I launch it. I need to click on "get started", accept the user agreement, then click next, then sign in or skip it, then click to add a printer, then find my printer, confirm it, then I have to skip the "what's new" window and then close the "release notes" window. EVERY TIME. 

Tested with a bunch of 5.x versions. 4.x versions does not have the problem, but my printer is not natively supported there.

### Expected results

I expect Cura to not pester me about logging in, or show me what's new and use the config file to remember that I have a printer already setup. 

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/20743270/cura.log)

HellAholic, 2025-06-17:
A) If you see an error message in Cura telling you about corrupted configuration (error message in bottom-middle of cura), it's possible that either the configuration for the 4.x is corrupted or you have a corrupted 5.x version configuration that is causing the issue.
So in the `~/.config/cura/` folder, remove all the 5.x version folders (as none of them work, it's less trouble to remove). Also make sure that the cura appimage has read/write permissions (either through chmod or manual through UI) to store the preferences.

B) If the problem still persists, also remove all the 4.x or prior versions so the `~/.config/cura/` is empty.

C) If the above steps did not work, manually adjust the cura.cfg in your configuration folder for the 5.10.1 to:
```
last_run_version = 5.10.1
accepted_user_agreement = True
```
This should prevent the what's new pop up from showing up.

wdiasjunior, 2025-06-18:
It seems I had to clean both the `~/.config/cura/` and `~/.local/share/cura/` directories to solve this.

Is there a way for me to back up my presets in bulk and transfer them to the new config directories? Or are the preset files not backwards/forwards compatible with different Cura versions?

closed 2025-07-03 by wdiasjunior