Ultimaker/Cura #20704: model fails to slice even after trying to fix with meshtools plugin

indubitablygregarious, 2025-06-18:
### Cura Version

5.10.1

### Operating System

Debian Linux

### Printer

Elegoo Neptune 3

### Describe your problem and add the package for technical support as a .zip here ⬇️

Not much to say, this fails to slice almost immediately

Thanks in advance!

[export-20250617-212411.zip](https://github.com/user-attachments/files/20786723/export-20250617-212411.zip)

GregValiant, 2025-06-18:
Thanks for the report.
I didn't have any trouble slicing the project file so I am unable to duplicate the problem (on Windows 10).

[image]

I used an on-line repair site and it gave this report:
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
--> 0 Non-Manifold edges  (?)
--> 0 Inverted normals  (?)
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 14122
--> Vertex count changed from 81789 to 82068 (**+279**)
--> Triangle count changed from 163672 to 164380 (**+708**)
Those are very minor problems and probably aren't the problem you are having.

After slicing fails, use the "Help | Export file for Technical Support" command and zip that file and post it here.  There might be a clue in the cura.log file.

indubitablygregarious, 2025-06-18:
Using formware.co I was able to fix the model and slice it

Feels kind of strange that just the one file is broken but I'm glad I was able to get this resolved

Thanks so much for your help, going to close this

closed 2025-06-18 by indubitablygregarious

GregValiant, 2025-06-18:
Formware.co will almost always make an adjustment to the number of vertices in a model.  If that is the only thing changed then I don't pay much attention to it.
I usually use Formware.co if I want a report on what errors might be in a model.  If not then I often use MS 3D Builder or PrusaSlicer repair.