Ultimaker/Cura #20714: Stop defaulting to dropping parts to build plate when ungrouping models for Multibody parts! 5.10.1

Nick2Percent, 2025-06-20:
### Is your feature request related to a problem?

Stop defaulting to dropping parts to build plate when ungrouping models for Multibody parts!

### Describe the solution you'd like

Just keep the initial settings that are applied to the model before ungrouping the model, and keep them from transforming during the process. 

ALSO!!!! Please keep files original origin location if possible. (Might just be amf file type issue idk.)

### Describe alternatives you've considered

NONE

### Affected users and/or printers

Myself.

HellAholic, 2025-06-20:
[image]
Go to preferences > Configure Cura... > General > uncheck the Automatically drop models to build plate.

Or before ungrouping: Select the group and uncheck the Drop to build plate for the group

[image]

If you still have the drop to build plate applied with these two steps, include an example file and submit a bug report.

Also consider that this is a free open source community software with people trying to help resolve issues free of charge. Being frustrated is understandable but this is not a place for you to vent out that frustration.

Nick2Percent, 2025-07-05:
> [image] Go to preferences > Configure Cura... > General > uncheck the Automatically drop models to build plate.
>
> Or before ungrouping: Select the group and uncheck the Drop to build plate for the [...]

__**Addressing the rude comment made by @HellAholic **__
_"Also consider that this is a free open source community software with people trying to help resolve issues free of charge. Being frustrated is understandable but this is not a place for you to vent out that frustration."_
Not trying to call you out here, but if I don't address this false assumption then it will set a negative tone for the conversation, please don't make assumptions beyond the face value of the text and words themselves. That being said I am mildly frustrated at that comment at the end of your message. To that I say the following First off everyone knows Cura is an open source slicer. And all of us who use it love it. Please understand that there was a link from a popup, on Ultimaker's website, that told me I cold make suggestions by clicking the link. That link brought me here. So if I am actually not allowed to make suggestions, which I do not believe is the case here, then perhaps someone should ask Ultimaker to remove the link requesting feedback from the website.  
Please don't try to make an improvement request seem like aggressive behavior; I am not saying that's what is going on here, because I am pretty sure its not; but it can be interoperated as shutting down other peoples opinions. If you look for negativity in a suggestion post you will be able to find it anywhere. Here is a link to an article that further explains the inaccuracies of assuming emotion based off of text alone. https://www.psychologicalscience.org/news/minds-business/youre-joking-detecting-sarcasm-in-emails-isnt-easy.html
Here is a link that has relevance for other reasons. https://www.tandfonline.com/doi/permissions/10.1080/0960085X.2022.2041372?scroll=top
To explain why my message didn't contain much helpful information.
I was trying to request the improvement in the best way I could. As I was typing it all out on my phone. Filling out the form in the google browser on the phone is difficult. I cant remember If I lost my paragraphs because I backed out of the web page when dropping my phone, or If I went in to select, copy, then paste certain things so I can structure my request better. Anyway I lost everything I typed, and was quite discouraged when I had to retype everything for the second time, so I tried to get my point across in as few words as possible. 
I just wanted to make the suggestion to people who know what they are doing. I cant do software development. Do not take it personally this is git hub, nobody is trying to start fights here... Well at least I hope not. Other forms out there I am sure can get pretty bad, I don't use Git much, but given that interaction I assume there is a big correlation in ambiguity, anger, frustration, and venting. . Infact this was the first time commenting on someone else's code or whatever this comment sections forum is. 
__Thankyou, your images made it easy to find the directory to turn off auto drop!__  __**This fixed the issue!**__ It would appear I wasn't finding the checkbox that needed to be unselected to prevent Cura from defaulting to automatically dropping the part down to the build plate G54 Z0 I assumed it was deleted with the addition of the boxes shown in figure 9001 _(Notice its OVER 9000!, and for the people who say that is was supposed to be over 8000. 9001 still counts. Link here:  
_https://en.wikipedia.org/wiki/It%27s_Over_9000!#:~:text=In%20response%20to%20Nappa's%20query,%22It's%20over%208000!%22._
  )_  I went to try the checkbox, but I must have missed it, and assumed, it was replaced with the per model setting. 

I also want to reply to "Or before ungrouping: Select the group and uncheck the Drop to build plate for the group" That was exactly what I was doing. Perhaps it was a glitch for that day, but After ungrouping the models It was automatically dropping the newly ungrouped models to the build plate against my wishes. I had the box unchecked prior to ungrouping. It would appear that I can not recreate the issue.

__**Figure 9001💯  :**__ 
[image] 

__**Addressing questions from @GregValiant** __
My apologies for not giving a prompt response back this account exists with my Burner Email, that of which gets so many notifications that I have had them turned off/ignore them. I do not have a project file to share. I can only explain the issue, as the parts I was working on were, and are not for public broadcast for IP reasons. Additionally I have not yet used the Project file feature, and look forward to it greatly.

**For anyone who was wondering what was going on on my end I will describe the issue the best I can for recreation the procedure was as follows:**

__**Recreation steps:**__
In preferences I had the check box that controls if the GUI automatically drops models to the build plate set to drop automatically. 
I had positioned my model into the most desirable orientation, and put my seam line where I wanted it. I then turned off drop model to build plate in the section shown in **_Figure 9001_** of which you can find in the section of this message addressed to @HellAholic. 
I then ungrouped the multi body part.
The result was the parts being automatically lowered, and the per part checkbox becoming rechecked for dropping the model to the build plate. This repeated for a while. Tried restarting Cura and all. Nothing would stop it.

GregValiant, 2025-07-05:
Not all "Model Export Utilities" are the same.  Blender uses a different Export App than (for example) AutoCAD does.  There might be user settable options in the Export Utility on the CAD side.
In addition, STL files have different properties than other file types like 3mf and OBJ.  STL files do not contain "location information" and can only contain a single mesh.  That mesh can be separated into it's constituent parts using the Mesh Tools "Split model into parts" command.  That will release the constituent models from their assembled locations and they will each be subject to the "Drop models to build plate" setting.

When I create a multiple mesh 3mf file and bring it into Cura, it can (not always and not often) "disassemble" itself and everything drops to the build plate.  When the model _disassembles on import_, the individual pieces arrange themselves at the middle of the build plate.  3mf files should not do that.  They should come in at the origin which (on my printer) is the left front corner.
The key there is "not all the time".  That makes it very difficult to chase down where the problem is within the code.  It's like trying to fix an intermittent electrical problem.  I have seen the problem that is described in this report.  I just can't make it happen on purpose.  On my system it just sort of randomly shows up.

I don't have the problem with STL files.  STL files "must" consist of a single model so if I happen to have multiple models they must be "merged" prior to exporting the file from CAD.  They never change position or "fall apart" on import.

BTW I've re-read your original report and HellAholic's comment and I don't see any rudeness in his response.  It seems to me that it was clear, concise, and to the point.

There are a lot of Cura users (100,000??).

[image]
I've been a volunteer here on the project for about 5 years and I have not seen any other reports of this behavior.  That doesn't mean there aren't other people with the problem, it just means that no one else has filed a report.  When you couple that with "intermittent and hard to find" it necessarily falls down the UltiMaker priority list below more pressing issues.

HellAholic, 2025-07-05:
1. Feature request does not begin with "Stop" in a 1 line sentence and certainly does not end with an exclamation mark. You're making a request, it should not be formatted as a demand. That is just English grammar, I did not assume, you just didn't provide any context other than a command.
2. If there is a request, it needs background information rather than 1 line. If your request is sincere, then please take the time to explain things clearly when making it. If you do not have the time or tools to create the request correctly, please do so at a later time when you're able. You can use tools such as ChatGPT or other LLMs to help format your request.
3. Affected users "myself", while true, it means what sort of workflows are affected here so it can be prioritized based on impact on users (Cura has 980K+ users that are connected to the internet just to give you some perspective). An example here would be "People that Export multi-body assemblies from `Application X` and want to print the parts individually for A,B,C,D reasons". This gives us an avenue to reproduce the issue or at least look at what sort of behavior we need to include/support.
4. If my message offended you in any way I apologize. Tagging someone 3 times in a single message however does not seem like a positive action.
5. You've failed to provide any example project files (3mf / amf /...) so still we cannot say whether there is an issue here or there is a need for a new feature.

Therefore, I do not see a feature request or a bug report that we can pursue here.

Nick2Percent, 2025-07-05:
> Not all "Model Export Utilities" are the same. Blender uses a different Export App than (for example) AutoCAD does. There might be user settable options in the Export Utility on the CAD side. In addition, STL files have different properties than other file types like 3mf and OBJ. STL files do not contain "location information" and can only contain a single mesh. That mesh can be separated into it's constituent parts using the Mesh Tools "Split model into parts" command. That will release the constituent models from their assembled locations and they will each be subject to the "Drop models to build plate" setting.
> 
> When I create a multiple mesh 3mf file and bring it into Cura, it can (not always and not often) "disassemble" itself and everything drops to the build plate. When the model _disassembles on import_, the individual pieces arrange themselves at the middle of the build plate. 3mf files should not do that. They should come in at the origin which (on my printer) is the left front corner. The key there is "not all the time". That makes it very difficult to chase down where the problem is within the code. It's like trying to fix an intermittent electrical problem. I have seen the problem that is described in this report. I just can't make it happen on purpose. On my system it just sort of randomly shows up.
> 
> I don't have the problem with STL files. STL files "must" consist of a single model so if I happen to have multiple models they must be "merged" prior to exporting the file from CAD. They never change position or "fall apart" on import.
> 
> BTW I've re-read your original report and HellAholic's comment and I don't see any rudeness in his response. It seems to me that it was clear, concise, and to the point.
> 
> There are a lot of Cura users (100,000??).
> 
> <img alt="Image" width="435" height="87" src="https://private-user-images.githubusercontent.com/64202104/462758571-2a2b1968-a8b1-40ee-88d2-a73394205472.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE3NDU5OTIsIm5iZiI6MTc1MTc0NTY5MiwicGF0aCI6Ii82NDIwMjEwNC80NjI3NTg1NzEtMmEyYjE5NjgtYThiMS00MGVlLTg4ZDItYTczMzk0MjA1NDcyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA1VDIwMDEzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzZTc1OTllM2NiZTQ5ZDMzNmNlY2NiMjJkMzVmODRkNDgyMDM2MGU1M2YwMDNkODc5MzYxMWJkMjhiZmI2NTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.THfaHy3vo7qvaHbKMyZ1XMuO8G1sX2BZkfJ7ByiaSOY">
> I've been a volunteer here on the project for about 5 years and I have not seen any other reports of this behavior. That doesn't mean there aren't other people with the problem, it just means that no one else has filed a report. When you couple that with "intermittent and hard to find" it necessarily falls down the UltiMaker priority list below more pressing issues.

Well, thanks for looking into it. If it comes up again, I understand that the priority for bug fixes are larger scale issues. I will be sure to come back with any new evidence if it happens again.

Nick2Percent, 2025-07-05:
> 1. Feature request does not begin with "Stop" in a 1 line sentence and certainly does not end with an exclamation mark. You're making a request, it should not be formatted as a demand. That is just [...]

Hello again. It appears as if you read some parts of my response and not others. Yes I agree the request I made was poorly formatted. I will not explain why as I already said so in great detail. I will, however, restate that I can not attach the files due to IP(Intellectual property) Reasons. If I get a chance in the next week or even today to make some models to figure out what is going on I will. 
When making the post subject I tried formatting it in a way that uses as little words as possible. Basically what I expect people to google when running into the issue. I am on my phone right now the UI is terrible I don't know how to exit the message and view the previous responses to verify the accuracy of what I am saying here, but I thought, “
Stop Dropping parts to Build-plate when ungrouping models.”
Was descriptive enough at the time. I have since changed it. Still have an Issue with it? Please let me know what you want it to say and I will edit it to do so. The exclamation point was because I was monotone when saying the issue. The exclamation points were to put emphasis on the words/statements that are not monotone. I love tacos! (Not angry) Exclamation points do not just mean yelling and anger. That being said the grammar I used was poor, and the subject having an exclamation point also seems poor practice And for that I am sorry.  
Moving on, When filling some of those question ticket things out i didn't really know what was being asked. I do not spend all day every day in git hub. I barely know how this application works. Now that you explain the structure of what you expect as responses to the question I feel I have a better understanding of what you are looking for, again I not a software developer. Additionally I only wanted to send one message instead of having to restate points over and over again. If you look at how the multiple @HellAholic was used it was so show who I was referring to instead of you having to reread everything, and gives you the option of only reading the part being said to you.  Implying that it’s negative/rude to address people. By their selected name seems a bit extreme to me. It’s just organization. I used the @ your name to keep it linked who I was referring to.  Incase you change your name or something.