Ultimaker/Cura #20720: Cura adding strange code to end of gcode file when using custom print settings profile

partyhatnat, 2025-06-21:
### Cura Version

5.10

### Operating System

Windows 11

### Printer

Bibo 2 Dual

### Reproduction steps

1. Open any stl file
2. Select my custom print setting profile 
3. Slice and save gcode
4. Print file 

### Actual results

Instead of the print head returning to it's home position and completing the print, it moves home and then readjusts itself to the middle of the print and stops.

Upon inspecting the gcode I found that it added some extra nonsense to the end. What should end with:
`;End GCode
M104 T0 S0                     ;extruder heater off
M104 T1 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
G91
G1 Z1 F100                                        ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-2 X-20 Y-20 F300 ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
M84                         ;steppers off
G90                         ;absolute positioning`

Instead tacks this block below the code above:

`M82 ;absolute extrusion mode
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = PLA Default\\nde
;SETTING_3 finition = bibo2_dual\\n\\n[metadata]\\ntype = quality_changes\\nqual
;SETTING_3 ity_type = draft\\nsetting_version = 25\\n\\n[values]\\nadhesion_type
;SETTING_3  = skirt\\nlayer_height = 0.3\\nsupport_enable = False\\nsupport_type
;SETTING_3  = buildplate\\n\\n", "extruder_quality": ["[general]\\nversion = 4\\
;SETTING_3 nname = PLA Default\\ndefinition = bibo2_dual\\n\\n[metadata]\\ntype 
;SETTING_3 = quality_changes\\nquality_type = draft\\nintent_category = default\
;SETTING_3 \nposition = 0\\nsetting_version = 25\\n\\n[values]\\ncool_fan_speed 
;SETTING_3 = 60\\ninfill_sparse_density = 5\\nironing_enabled = False\\nmaterial
;SETTING_3 _flow = 105.0\\nmaterial_print_temperature = 195\\nskirt_height = 1\\
;SETTING_3 nspeed_print = 70\\nsupport_brim_width = 0\\nsupport_infill_rate = 10
;SETTING_3 \\nsupport_z_distance = 0.2\\ntop_bottom_thickness = 1.2\\nwall_thick
;SETTING_3 ness = 1.2\\nz_seam_type = random\\n\\n", "[general]\\nversion = 4\\n
;SETTING_3 name = PLA Default\\ndefinition = bibo2_dual\\n\\n[metadata]\\ntype =
;SETTING_3  quality_changes\\nquality_type = draft\\nintent_category = default\\
;SETTING_3 nposition = 1\\nsetting_version = 25\\n\\n[values]\\ncool_fan_speed =
;SETTING_3  60\\ninfill_sparse_density = 10\\nironing_enabled = True\\nmaterial_
;SETTING_3 flow = 105\\nmaterial_print_temperature = 195\\nskirt_height = 1\\nsp
;SETTING_3 eed_print = 55\\n\\n"]}`

### Expected results

Expected the following code to be the end of the file, with nothing following, as per my printer profile settings:

`;End GCode
M104 T0 S0                     ;extruder heater off
M104 T1 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
G91
G1 Z1 F100                                        ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-2 X-20 Y-20 F300 ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
M84                         ;steppers off
G90                         ;absolute positioning`

### Add your .zip and screenshots here ⬇️

[june21_test.zip](https://github.com/user-attachments/files/20846635/june21_test.zip)

partyhatnat, 2025-06-21:
Thank you for your response and explaining the contents of that commented code. I had never noticed that "paragraph" until I installed ver 5.10, and I only noticed it because my printer wasn't returning home at the end of the print following that update. Removing that comment block seemed to fix the problem so I thought it was a bug. I suppose the timing was just a coincidence. 

You may disregard this, then. Thanks again.

GregValiant, 2025-06-21:
That paragraph has no effect on what the printer is doing.  I looked at the pull requests here and it appears that the BIBO definition file has not been altered in quite a while.

At the end of your Ending G-Code is the "G28 X0 Y0" line which should park your print head out of the way at the Auto-Home location.
If it isn't going to that "Minimum X and Minimum Y" location, then something isn't acting right.  

In some firmware G28 X Y is the "Auto-Home the X and Y axes" command and G28 X0 Y0 is "Auto-Home if the location has been lost" as would happen if the stepper motors disarmed.  You have the M84 line (disarm steppers) below the G28 so that should not be a problem.  If the G28 X0 Y0 is being ignored because the steppers know where they are, then the print head would likely stay at X-20 Y-20 as the previous command stated.

I'll go ahead and close this.  You can still comment, but this isn't a "bug problem".

closed 2025-06-21 by GregValiant