Ultimaker/Cura #20727: Stange thin supports

alainbellino, 2025-06-25:
### Cura Version

5.9.0 and before

### Operating System

Ubuntu Linux

### Printer

Creality Ender 3 S1

### Reproduction steps

1.I sliced ​​an stl file in cura1. 

### Actual results

I got a file with normal organic supports but also very thin unusual supports (see photos)

### Expected results

I want to know what are these "supports" and how to remove them.I specify that they are printed

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
[image]

[CE3S1_tete en 2 parties 1.zip](https://github.com/user-attachments/files/20905426/CE3S1_tete.en.2.parties.1.zip)

HellAholic, 2025-06-25:
Your model is not watertight.
https://support.ultimaker.com/s/article/1633018228456

[image]

alainbellino, 2025-06-26:
Thanks , i used only 3D Print Tool box, i will  use mesh repair on Blender and mesh tool on cura and probably come back on modeling for some parts because my models are complex and have many defects, until now it did not pose too many problems in printing but I will try to do better.
I didn't think these abnormal rods/supports could come from the model's defects, I thought they came from using the old addon for manual supports, they appeared at that time but they still exist despite the removal of the addon.

Thanks a lot

HellAholic, 2025-06-26:
So the most reliable way to handle them for complex models is by doing it the hard way (manually and slowly). Automated methods (mesh repair on blender for example) are more like hammers, the might get the job done but they will also wreck all the fine details.
The engine usually resolves small issues (using a smaller hammer) so a lot of times you don't even see any artifacts on the print.

There are couple of things at play, one part is the model, second part is the profile, third part is the slicer. If I just revert all your profile changes, then those abnormal rods/supports almost disappear (probably few will remain but get removed in the engine with the small hammer), and the support generate as normal-ish, so you could tweak the settings to see which one would allow you to slice without those supports/rods. Example, try increasing the `Branch Diameter Angle` and `Preferred Branch Angle`

[image]

Another solution given the complexity of your model could be something like the `Resin2FDM Lite` plugin for blender (resin support for fdm) then you can manually place the supports and don't rely on the generated supports. But this is mostly used for miniatures and with 0.2 -0.25mm nozzles, but give it a look, it might help you in some way.

I do see less of the rods/supports after reorienting the model and tweaking some settings.
As an example project below has model reoriented and increased the `Preferred Branch Angle` and `Maximum Branch Angle`
[CE3S1_tete en 2 parties 1rectifié_example .zip](https://github.com/user-attachments/files/20932851/CE3S1_tete.en.2.parties.1rectifie_example.zip)

[image]

So the above are suggestions for the paths that might resolve your issue but the recommendation would be fixing the model first and then attempting to print.
It's a nice model and a long print, so take your time and see what works for you.

alainbellino, 2025-06-27:
Thank you very much for your time and the detailed explanations, which have opened up some avenues for me.

My models are indeed complex and obtained using methods that I consider relatively marginal, such as the systematic use of the shrinkwrap modifier (which sometimes creates very distorted meshes) or the interlacing of all the objects used. At first, I wasn't sure I'd be able to print my models, and I didn't always take these flaws into account when the prints were successful.
In this case I have problems with  layers shift that I may have attributed to these thin supports, but it could also be that this comes from defects in the model and therefore small surpluses of material due to printing errors.

-I've already played with the angle settings, but I'm going to dig deeper.

-I'm going to try Resin2FDM, or maybe even the new Cura add-on for manual supports, which I haven't installed yet.

-The orientation is more complicated since I'm trying to maintain a horizontal orientation so that the print lines are horizontal on all parts of my model. But I'll take a closer look at the angle settings on the project attached in your comment.

 -Finally, I'm a novice at printing and modeling, and I also have to manage foundry parameters because my models are intended to be cast in lost wax (resin). 
I'm a sculptor, and this model is inspired by a completely handcrafted creation, photos of which I've included below.
so you will see that your comments are very useful to me

Once again, thank you very very much.

[image]
[image]
[image]
[image]

HellAholic, 2025-06-27:
My pleasure, glad it was helpful.
If you have questions or follow ups you can always start a discussion if you're not sure if it's a bug or just something that could be related to model / settings / etc.
I'll convert this report to a discussion so others can also participate and view (@GregValiant might be interested) and re-open it, so feel free to continue on this one.

GregValiant, 2025-06-27:
Wow.  That is a really nice model.
I have found that the problem with most "Really Nice Models" is they often have features that are too fine for FDM printing.  That's what I see there.  Scaling a model down to a printable size is a problem because fine features will become un-printable.

Is the original work welded construction?  It appears to be constructed of some industrial looking stuff.  Is that armored electric cable and a round junction box in there?

[image]

[image]

Consider printing a model of a house.  The full size house has a shingled roof, mullions in the windows and a wrought iron fence around the front porch.  At 100% scale the house and all it's features would be printable.  Once you start scaling it down the house looses definition as individual features become too fine to print.  The window mullions, shingles, and railing would disappear.  They become too fine to either slice or print.  When that is the case, the user can ignore the problem, or adjust the model so the features become printable.  More "Fun Scale" than a "True Scale".

If you were to close up all the little gaps in the model, and fill the interior while leaving your mounting tabs and things, this would have a much better chance of holding it's shape and printing, it may not be quite the same.  That's the trade-off when converting "artsy craftsy" to "Industrial Art".

On the [UM Forum](https://community.ultimaker.com/discover/) there were a couple of members that did artsy craftsy very well.  Unfortunately, ["CloakFiend"](https://community.ultimaker.com/profile/331687-cloakfiend/) and ["kmanstudios"](https://community.ultimaker.com/profile/350891-kmanstudios/) don't come around anymore.  You might find some interesting tips if you look up a couple of their posts regarding things they made.  CloakFiend was into "plating" models and did some "Lost PLA" casting.  

There have been a couple of incredibly detailed models of the Eiffel Tower that have shown up here.  Printing them...I don't know.  A print failure after 30 hours would be a heartbreaker.

GregValiant, 2025-06-28:
So many reports come in here regarding models that really have no interest to me (I was sick of Mandalorian helmets and Crystal Snakes!!) that it is a pleasure to see something different and to actually have to think for a change.

There are a couple of settings that don't get used much that you might want to try.  In the Experimental Section is "Make overhangs printable".  It will add chamfers to the bottoms of features so that (in many cases) support would not be required.  Those chamfers do alter the design so it may not be something you can use.

The second is the Mesh Fixes "Remove All Holes".  In the case of this model, it didn't appear to be of much use, but closing some of the side holes in the model might make it worthwhile.

alainbellino, 2025-06-28:
I know these two functions but I haven't really used them. The remove holes function is probably useful to correct mesh defects still present but the goal is to remove them during modeling. However, as you say, I will keep it in mind because it can be useful in certain cases. The make overhangs printable function is interesting, I will look into it a little more, unfortunately at 60° (minimum overhang angle) there are no more supports at all and in this specific case there are islands on the lower face which is no longer supported so manual supports must be placed. 
 I will try custom supports reborn I tried to re-slice the .mtl project but cura still does not consider the bottom of the model as overhanging. At 50°, I have supports.

I have another question, if it's not too much trouble:
I have a strange support that's grabbing a part of the model that could be supported by the support next to it. I tried modeling an internal support (square tube), and nothing works; the support is still there. Why ? I suspect it's causing the layer shifts because it's long, very sloping, and therefore a bit flexible.

[image]

Regarding the Mandalorian helmets, I made a Darth Vader and an R2D2, here are the photos. I haven't exploited this vein, but it's not impossible that I'll make some in digital 3D If I manage to optimize the process. 

[image]

[image]

The Darth Vader was purchased by George Lucas himself and should be in the new Museum of Narrative Arts in L.A., which should open next year. In this case subject isn't important; what matters is the implementation.

GregValiant, 2025-06-28:
That long narrow support seems to be a function of the angle of the model to the build plate.  When I slice the project that HellaHolic posted that support leg doesn't show up.

This is a very difficult print to get right.  You have holes in the model, The walls vary in thickness, the supports are difficult to add and may be a SOB to remove.  I think I would want a dual extruder printer with water soluble supports for this.  Along with the chance of a regular ole' print failure hours into it, there is the chance of hurting the model when removing all those supports.  Some of the supports will be stronger than the thin wall area of the model that they are stuck to.

Sometimes it's advantageous to print the model in pieces, and glue them together.  I subtract 2.2mm diameter cylinders from the model at the "cut" line and then split the model (MS 3D Builder is good at that).  At assembly I glue in pieces of filament as locating pins.  The halves go together perfectly no matter their shape because the pin holes are perfectly located in relation to each other.

You have your work cut out.

alainbellino, 2025-06-29:
It doesn't appear because the part has been reoriented, but I'll consider all the advice in this discussion and probably find a solution to print this model.

Support removal is also made more difficult by using transparent filament (without pigments) due to the subsequent casting step.
but I get there with a little patience and suitable pliers

Soluble supports require soaking in water, which I'm wary of because the moisture absorbed by the PLA is likely to cause problems in the casting process. I want to avoid using solvent-soluble filaments, especially for large parts.

This object is already cut. I can make more pieces, but that increases the difficulty of soldering and removing joint defects in bronze (and even almost more so on PLA due to its transparency). The texture is complex, and what is easy on a smooth part is more complicated on a highly textured one. That said, assembly before casting is also an option if I can find a castable adhesive.

Thank you for the assembly technique with pieces of filament, it will certainly be part of my thinking, I remain hopeful of finding the right solutions, I will digest all the advice given here, thank you very much. I will come back here to give feedback in some time.