Ultimaker/Cura #20735: "Reload All Models" doesn't work for multiplied models

rondlh, 2025-06-27:
### Cura Version

5.10.1

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

1. Open a STL model
2. "Multiply Selected" to create more of the same models
3. Change the opened STL file
4. "Reload All Models"

Now only the first model is reloaded, all copies are not.

### Actual results

Only the first model is reloaded, not the copies

### Expected results

All Models should be reloaded, including the copies

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-06-27:
Thanks for the report.
I can duplicate this problem.  I seem to recall that this bug has been reported previously.

When Cura notices that a model file has changed and this dialog comes up...

[image]

clicking on "Reload" seems to reload all the models correctly.

When the "Reload all Models" command is chosen from the Right-Click menu, there seems to be a bug.

The Cura team will take a look.

rondlh, 2025-06-28:
> When Cura notices that a model file has changed and this dialog comes up...

I don't get this reload dialog. I "change" the model by renaming the file, not by overwriting the file, perhaps that is not causing the dialog to trigger.

Thanks

HellAholic, 2025-06-28:
He Greg, I was looking at a community PR that would add some functionality related to reloading models and I also saw some issues, haven't looked in detail yet it is on the todo list. I'll take a closer look in combination with that PR. Thanks for flagging it.