Ultimaker/Cura #20746: Kalypso Logo Slicing Failed

gypsy-cat, 2025-07-06:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Anycubic Kobra Plus

### Describe your problem and add the package for technical support as a .zip here ⬇️

I am trying to slice a model that fits well within the area of my build plate, but it keeps failing until I size it down to 86%.

[export-20250705-214727.zip](https://github.com/user-attachments/files/21084212/export-20250705-214727.zip)

gypsy-cat, 2025-07-06:
> Hey @gypsy-cat <https://github.com/gypsy-cat> , you have the Valve plugin installed. The plugin has a limit on the size of the gcode it can handle (larger model, larger gcode). Disable the plugin and it should slice at 100% size.

Where do I find the installed plug-ins to uninstall them?  

Also is there a way to compute the amount of material a build will use?

 

Best Regards, 

 

Suzie Fuente 

4215 Wildwood Road 

Moss Point, MS 39562 

Cell:  228-369-9556 

Home:  228-202-7900

HellAholic, 2025-07-07:
You're welcome. I'll go ahead and close this since the issue has been resolved.

closed 2025-07-07 by HellAholic