Ultimaker/Cura #20756: [5.9.0] Pause at height - misleading option for M109

IronicChloride, 2025-07-09:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Irrelevant

### Reproduction steps

Add Pause at height to post-processing.
Select "Use M109 for standby temperature?"

### Actual results

The printer goes to standby temperature as expected. M104 is used (correctly) to set the temperature, despite what that option says.
After resuming the temperature is set to the printing temperature using M109 (but that's not what that option says).
The popup tool tip describes the action correctly.

"Use M109 for standby temperature?" checked
```
M104 S0 ; standby temperature
M117 interface C flange
M84 S3600 ; Keep steppers engaged for 1h
M300 S440 P1000 ; Beep
M25 ; Do the actual pause
M109 S200 ; WAIT for resume temperature
```

"Use M109 for standby temperature?" unchecked
```
M104 S0 ; standby temperature
M117 Interface bayonet
M84 S3600 ; Keep steppers engaged for 1h
M300 S440 P1000 ; Beep
M25 ; Do the actual pause
M104 S200 ; resume temperature
```

### Expected results

The problem is ill-posed. Following the logic, the standby temperature should be set using M109. Which is expected but not desired. The option title should be changed.

### Add your .zip and screenshots here ⬇️

Change the option title to "Use M109 to resume from standby"

GregValiant, 2025-07-10:
Thanks for the report.
M109 is used for the resume temperature only.  The tooltip appears to be in error.

This is the current PauseAtHeight with the verbiage change you asked for.
[PauseAtHeight.zip](https://github.com/user-attachments/files/21170685/PauseAtHeight.zip)

I have a full re-write waiting to be submitted but the Cura team is currently busy with other things.  It will be a while before the change is made.  This is what it looks like.  (If the "Reason for Pause" is "All Others" then the filament change settings are hidden because they aren't needed.)

[image]

IronicChloride, 2025-07-10:
Thanks for the update on the upcoming change.
As for the current plugin, I've figured it out the hard way, when the printer resumed at 170°C (that's what's in the firmware), should have read the tool tip, maybe I would then at least checked the G-code.
In case anyone else is bothered by this and ends up here, by default the script sits (on Windows) in C:\Program Files\UltiMaker Cura 5.9.0\share\cura\plugins\PostProcessingPlugin\scripts
Download the provided .zip and extract the PauseAtHeigth.py to this directory (you can safely overwrite the existing script).

GregValiant, 2025-07-11:
_"...the printer resumed at 170°C (that's what's in the firmware)..."_
The firmware shouldn't have anything to do with the temperature itself.  The Cura material might have 170° as the Standby Temperature but that isn't called by the current script.  The default Standby Temperature of the script is "0".   In the new script it will call for the Material Standby Temperature to be the default.

IronicChloride, 2025-07-11:
I have a custom firmware with some pause features configured. I was using M25 for pause. It waits for the user to press the button and then heats the hot end, allows an option to purge and then continue to print. From what I see, the 170°C is EXTRUDE_MINTEMP in Marlin firmware, that's what the printer sets after resuming from pause since it the has to extrude some filament. When paused the temperature set point was 0°C.

GregValiant, 2025-07-11:
_"I have a custom firmware..."_
Well - there you have it.  There is no way to tell what any individual may have changed in their firmware.  When writing these scripts certain things must be assumed.  Custom macros and other personal things can/will cause unexpected behavior.  

In a pinch you might have to enable "Search and Replace" to fix or adapt things that another built-in script might do that doesn't work with your machine.  IDEX machines in particular cause problems because virtually every one of them works differently than the others.  It isn't that Cura can't be made to work with them but rather that they require a custom fix.  For example Flash Forge printers do not use G0 as a movement command.  Everything is G1.  Consequently Flash Forge users must slice with Flash Print unless the gcode is heavily post-processed.

This is the PauseAtHeight version I will propose.  It also has unload/reload/purge options.  It might not work at all with your printer.  You can play with it and see.  I have changed the name of this file to "PauseAtHeightAlpha.py" so you can install it alongside the stock version.  If you put it into the "scripts" sub-folder in your "configuration folder" it will be available in Cura.

[PauseAtHeightAlpha.zip](https://github.com/user-attachments/files/21187662/PauseAtHeightAlpha.zip)

You can suggest changes (in this thread) but keep in mind that the script must work with "all printers" as much as that is possible.  "Model specific commands" won't be added.

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant