Ultimaker/Cura #20759: Updated "One at a time" printing

o6uipi, 2025-07-10:
### Is your feature request related to a problem?

I realized when using the "one at a time" setting that it automatically reduces the build space when I add a second object.  I feel like this makes the one at a time feature useless unless it's a bunch of small objects.  The software already allows us to input the dimensions of our hot end and gantry so I think they could add at least one more dimension (gantry width) to allow actual multi-item printing.

### Describe the solution you'd like

I think it’s a bit dumb that it assumes the entire build plate is gantry height as soon as you put two items on the bed when they could simply use the same highlighted interference area for the X and Y axis but extend another section all the way left and right. But I guess that would only be the case for specific styles like CoreXY printers so there would need to be an option to identify which type it is.

### Describe alternatives you've considered

I went in my settings and put an incorrect gantry height and then manually positioned my parts to be out of the way of the gantry when it changed to the new location

### Affected users and/or printers

this affects everyone technically since it's part of the software design

### Additional information & file uploads

here's what that would look like. you place an object and it highlights the gantry interference area so the printer knows to z hop over it when crossing.  The complexity could also be increased by calculating the height of each section of the part so that it shows where each section would interfere, like in my case how the tall center part will interfere but the lower outside area will not so the interference area can be shrunk down.

[image]

GregValiant, 2025-07-11:
This has been requested several times previously.
Yes, you can do this manually.  Trying to do it automatically just isn't in the cards as there _**IS**_ an interference point there and you would be asking the software to ignore it.  Make no mistake - somebody would absolutely smash into a previously printed part.

The way the code is written; that crash would be the users fault.  If someone were to change the code then the crash would become Cura's fault.

This request isn't a terrible idea, but I don't see it happening.  Those previous requests ended up in the "Won't Do" bin.

o6uipi, 2025-07-11:
I don't see how that would still make an interference point if you have a setting to put in the gantry dimensions. The software already knows the gantry height from the bed zero so all it needs is the rest of the gantry shape and it can calculate an interference zone for whatever axis the gantry runs on.  

For example, my gantry is a 20x20 extrusion and runs the X axis so if Cura had settings to input that info it can highlight the interference area like the picture I posted.  If the part ends up being taller than the gantry and in the way of the new part position then the software can Z hop over the already complete part if it needs the gantry to cross over it.

HellAholic, 2025-07-11:
You're focusing on your own printer and your specific print job, but this feature has to work across all printer types, not just one.

Let me expand on why this won’t be implemented:

1. **Physical Safety Risks Across Variants:**
Imagine a printer with an enclosed chamber. The first part prints to full height (e.g., 30 cm), then the bed is raised to start the next part. If the software doesn't enforce a full-height interference zone, it could raise the print bed and crash the finished part into the top of the enclosure or the gantry itself. This can seriously damage the printer or the part. There’s no way for the slicer to reliably account for such mechanical differences across all hardware.

2. **Too Many Hardware Configurations to Support Safely:**
The variety of printer designs (CoreXY, Cartesian, delta, bed-slinger, IDEX, belt printers, etc.) means that implementing a motion-aware collision system would require an unmanageable number of kinematic profiles and assumptions. Most slicers are designed to be printer-agnostic, and trying to accommodate these differences would be a maintenance nightmare and source of constant bugs.

3. **Incomplete or Inaccurate User Printer Profiles:**
Many users either don't configure their machine profiles correctly or use generic ones. If the slicer allowed behavior based on inaccurate gantry dimensions or unsupported motion types, it would introduce real-world collision risks—especially for beginners who don’t understand the implications.

4. **Z-Clearance Can't Be Assumed Just from Motion Type:**
Even if you identify a printer as CoreXY or bed-slinger, that doesn’t mean the gantry or nozzle can safely pass over a partially printed object. There could be wiring, fans, mounts, or structural elements that interfere. The safe default is to assume full vertical clearance is needed.

5. **Edge Cases and User Error Would Cause Major Damage:**
Any bug, oversight, or mis-configured setting could result in catastrophic collisions between the gantry and a finished part. For a feature that only benefits a small subset of use cases, the risk is not worth it.

6. **Manual Overrides Exist for Advanced Users:**
As you mentioned, advanced users can already bypass the restriction by faking gantry height and positioning parts manually. That flexibility already exists for users who understand the risks. But enabling this kind of behavior by default or making it more accessible would lead to more broken printers than successful prints.

In short, this isn't about lacking the will to implement a smarter system—it's that no implementation could reliably and safely work for the massive variety of printers and use cases out there. The current approach is a compromise that protects users and equipment.

GregValiant, 2025-07-11:
@o6uipi if you add a second instance of your printer and then customize the "Printer Name", the "Print Head" dimensions, and the "Gantry Height" then you can select that printer when you need to print two tall models in one-at-a-time mode.

I'll mark this as a "Won't fix/do" and go ahead and close it.  I hope you understand.

closed 2025-07-11 by GregValiant