Ultimaker/Cura #20761: Failed

marvinped, 2025-07-11:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Pro

### Describe your problem and add the package for technical support as a .zip here ⬇️

[export-20250711-200806.zip](https://github.com/user-attachments/files/21191309/export-20250711-200806.zip)

marvinped, 2025-07-11:
Carreguei o arquivo e o fatiamento falhou.

GregValiant, 2025-07-12:
Thanks for the report.
I didn't have any trouble slicing the file on my Win10 system.
From your log file is _"Backend exited abnormally with return code 3221226356"_.
Maybe @HellAholic will be able to take a look.

HellAholic, 2025-07-12:
Seems like a calculation error when generating the tree support.
Change the Support Placement to touching the build plate and see if that works.
You can also move the model slightly to the right or to the left on the build plate to see if that helps.
I can't reproduce the issue locally on Win 11
[image]