Ultimaker/Cura #20771: Title bar prevent from normal use on GNOME DE wayland.

chm0d777, 2025-07-18:
### Cura Version

5.10.1

### Operating System

Arch Linux, GNOME, wayland, linux-zen.

### Printer

Elegoo neptune 3 pro (irrelevant)

### Reproduction steps

1. install archlinux with gnome DE.
2. install dash to panel extension (make issue worse)
3. yay -S cura-bin
4. open cura

### Actual results

title bar doesn't dissapear and preventing from normal use even if i try to maximise window
[image]
workaround:
wmctrl -r "Ultimaker Cura" -b add,fullscreen 

### Expected results

No title bar on top.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-07-18:
Thanks for the report.
This might not have anything to do with Cura.
Are you talking about the toolbar along the bottom?  The "TitleBar" is at the top and says "UltiMaker Cura".

Does that toolbar along the bottom have "Settings" or "Properties" when you right-click on it?
(The Windows Taskbar gets in the way unless it is set to "AutoHide".)

HellAholic, 2025-07-18:
Your screen resolution is the minimum setting in which Cura is still usable (720P - mostly the height is important). In that resolution, you can use Cura but you would have auto-hide for the taskbar for the Slice button and the other components to be fully visible. Increase your screen resolution to 800P if you don't want to use the auto-hide taskbar.
I don't see a bug here.

chm0d777, 2025-07-19:
Title bar at the top that says "UtilMaker Cura" shouldn't be rendering because it's non gtk app, it's taking free space and preventing from using bottom buttons. Toolbar along the bottom (dash to panel) have settings/properties, but there is no autohide. The issue doesn’t seem to be directly caused by Cura itself, but rather by how window decorations are handled in gnome under wayland.
Cura appears to use qt and draws its own title bar using client-side decorations, which gnome doesn’t override or modify. That title bar takes up space and blocks interaction with the bottom part of the UI on smaller screens.
I was trying to find a way to hide or customize this TitleBar, but it seems that GNOME currently doesn’t allow that unless the application supports it directly. I'm fixing it using bash script that execute command "wmctrl -r "Ultimaker Cura" -b add,fullscreen" when cura is fully loaded.

chm0d777, 2025-07-19:
> I don't see a bug here.

Ok sure, i thought this was a bug, because non gtk app shouldn't render title bar. Anyway, it can be fixed with custom `.desktop` file that runs a bash script to open Cura, waits for it to load and executes `wmctrl -r "Ultimaker Cura" -b add,fullscreen`