Ultimaker/Cura #20782: Slicing Issue

OniNoDeibbido, 2025-07-21:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Neptune 4 Max

### Reproduction steps

Slicing a file

### Actual results

I receive a slicing failed notification

### Expected results

The file would slice

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/21353553/cura.log)

[image]

I cannot load the Project File, I keep getting an error that says  "

GregValiant, 2025-07-22:
If the project file is too large then Github won't allow it.  You can try posting it to Google Drive or another site.  We must have permission to access the site in order to download the file.

I have seen problems that occur when a model is multiplied.  One of more of the models will have errors, and they keep Cura from slicing.  You can check each of the duplicates to see if then have errors.

OniNoDeibbido, 2025-07-23:
That is good to know, because I have duplicated items multiple times. How
can I check each item to see if there is a problem?

GregValiant, 2025-07-24:
Go to the Cura MarketPlace (button in the top right corner) and install the "Mesh Tools" plugin.  That will add tools to the right-click menu.  With a model active select "Mesh Tools" and then "Check Mesh" and you will get a report on the active part.
In Preview, sometimes "X-Ray View" will show the errors.
Here is a screenshot of Mesh Tools.  There are also some settings under the "Extension" menu in Cura.

For repairing individual models I have found that MS 3D Builder is pretty good if you can still download it from the Microsoft Store.
I also use [formware.co.OnLineSTLRepair](https://formware.co/OnlineStlRepair) but for your situation Cura's Mesh Tools is a better solution.

[image]