Ultimaker/Cura #20792: enabling wipe nozzle between layers causes weird behaviour

cletero, 2025-07-24:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Creaality ender 3 Pro

### Reproduction steps

1. Load model
2. Move model off center
3. Check that travel moves can be seen in PREVIEW tab
4. Use my usual slicing settings, but enable "Wipe Nozzle Between Layers"

### Actual results

When "Wipe Nozzle Between Layers", Cura moves the nozzle all the way to the center of the printer (at every layer change apparently, but not 100% sure) and this causes print time to go up and probably gives the nozzle more time to ooze, etc.
The included jpg shows the difference in travel moves between wipe enabled vs no wipe enabled.

### Expected results

There should be no difference in travel moves when "Wipe Nozzle Between Layers" is enabled.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2025-07-24:
Why would you expect nothing to change when you enable a setting?

_Wipe Nozzle Between Layers_ is meant to perform a travel move at the start of every layer, with the intent being to wipe the nozzle on something like a brush. In other words, it's going to increase print times because it's performing a movement that it otherwise wouldn't be.

Its default configuration is more safe than practical, but there are a lot of settings available (which you don't have shown) to fine tune it - speed, Z hop, whether it retracts or not, how many times it should wipe, how long each wipe should be, and importantly the X value where your brush is located. There is no Y configuration, which makes it better suited to bed slinger/Mendel-style printers, but there are ways to make it work on other motion systems (such as having the brush attached to the X gantry or a full-length brush).

I can only assume you're confusing this for another setting - perhaps the _Outer Wall Wipe Distance_ or _Infill Wipe Distance_ settings, which is a much shorter movement at the end of their respective line type that. These will increase print times but generally not by much, won't move outside of the model and attempt to pass over already-printed filament.

GregValiant, 2025-07-24:
@cletero This seems to be "Expected Behavior" (although it appears that "Wipe between layers" isn't configured correctly for your printer.)

Does @Asterchades description explain what is going on?  I'm inclined to remove the "bug" label and close this.

closed 2025-07-24 by GregValiant