Ultimaker/Cura #20793: Printing temperature

Dengeplus, 2025-07-24:
### Cura Version

5.10.1

### Operating System

MacOS 15.5 

### Printer

Ultimaker 3

### Reproduction steps

Testing the quality of a print over a printing temperature range in increments of 5 °C

### Actual results

The prints all look the same.  Followed the temperature graph from the UM3 API and noticed the temperature setpoint is always set to 200 °C regardless of the printing temperature set in Cura.

### Expected results

when setting the printing temperature in cura to 180 °C the G-code command should be M104 S180 and M109 S180 instead of M104 S200 and M109 S200

### Add your .zip and screenshots here ⬇️

[image]

[image]

GregValiant, 2025-07-24:
Thanks for the report.
I can't duplicate this.  When I set the E1 temperature to 180 the print starts at 180.  It can drop down to "Final Printing Temperature" over the course of the last few layers.
If you have any post-processors running to adjust the temperature they can have a definite effect.

Please provide a project file with the setup that you find in error.

Dengeplus, 2025-07-24:
Hello Greg,

On what printer did you try to duplicate the problem?

What I found is that regardless what figure I fill in Printing temperature the values behind M104 and M109 stay S200!  So the temperature of the core doesn’t change…
What do you mean with "When I set the E1 temperature to 180 the print starts at 180”?

I’m only using Cura to print and have no post-processors running.

Sincerely,

Hans

GregValiant
 left a comment 
(Ultimaker/Cura#20793)

GregValiant, 2025-07-24:
I used a UM 3 as the printer.  Both extruders were enabled.  "M104" appeared in the G-Code 106 times.  It starts at S180 and finishes at S170.

I still don't have a project file.  There is a reason we ask for them.

Dengeplus, 2025-07-25:
Hi Greg,

Indeed it is for the temperature problem.

[mrl_zpproj.3mf.zip](https://github.com/user-attachments/files/21427443/mrl_zpproj.3mf.zip)

Hope this is the correct file!

Greetings.

closed 2025-07-25 by GregValiant

Dengeplus, 2025-07-25:
Hi Both,

Ok, looks good News.  I have the impression I overlooked something by a lack of knowledge :-(
From the comment of HellAholic and from you I suppoosten I have to click on the 2-tab at the top to assign the model to the second extruder (T1), is this a correct reasoning?

Thanks for the help!

Greetings