Ultimaker/Cura #20802: 5.10 and 5.10.1 fatal error on start

LilBub, 2025-07-27:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Ultimaker 2+

### Reproduction steps

Started Cura 5.10 and immediate crash, prompt to send report
Decided to upgrade to 5.10.1 and same issue

Cura had run fine on this machine before.

### Actual results

Crash Report, program fails

### Expected results

### Add your .zip and screenshots here ⬇️

[image]

LilBub, 2025-07-27:
[cura.zip](https://github.com/user-attachments/files/21451001/cura.zip)

HellAholic, 2025-07-27:
Close Cura, Clear out (delete) the local cache folders of cura in `%localappdata%/cura` and then open Cura again.
If that still fails your configuration is corrupted and you might need to manually remove the corrupted files or clear out the configuration folder.