Ultimaker/Cura #20804: Loading Model Failed w? all models on Thingiverse.com

MandoJay, 2025-07-27:
### Cura Version

5.10.0

### Operating System

Windows11

### Printer

Anycubic Kobra Plus

### Describe your problem and add the package for technical support as a .zip here ⬇️

trying to open STL files on thingiverse.com using the "Open in Cura" option, something I've done dozens of times before, and all of the sudden it keeps saying "Loading Model Failed" when I try to use the "Open in Cura" option. This isn't isolated to one particular file, as it does it in any I try to open. Files seem to open normally if I first download the file and then use the "Open with..." option and then select Cura. This has never been the case before, and I shouldn't have to download every file, as I have never had to previously! This has started happening in the past week or 2 and is a constant issue. Seems to have started when I downloaded the newest update to Cura. I tried going back to the previous version with no luck, I've tried uninstalling and reinstalling multiple times. I have already tried disabling the USB Printing plugin, it didn't help and am still having the issue. I know that others have had this issue, but I have seen no concrete solution to this problem, PLEASE ADVISE! 
***(If you ask for the Log File, please advise how much of it you want. I'm happy to C&P it, but it is a HUGE amount of info so please advise if you want the entire list or just from today or how much of it that you need!)*****

MandoJay, 2025-07-27:
This is NOT from a project that I am creating, it is from any file on thingiverse.com. It fails to load on anything I try to open, and I know for certain that every single file on thingiverse.com isn't missing a .zip file. Especially since downloading the file first and then using Ultimaker Cura to open the file works just fine (as well as opening them in Anycubic Slicer), it is only the "Open in Cura" on the website that fails to load. I am not the only one to have this problem, and it has nothing to do with creating my own project. BAD BOT!

MandoJay, 2025-07-27:
[export-20250727-004422.zip](https://github.com/user-attachments/files/21451840/export-20250727-004422.zip)

MandoJay, 2025-07-27:
[export-20250727-004835.zip](https://github.com/user-attachments/files/21451851/export-20250727-004835.zip)

MandoJay, 2025-07-27:
[export-20250727-004942.zip](https://github.com/user-attachments/files/21451854/export-20250727-004942.zip)

PartySausage, 2025-07-27:
If you've recently updated your version of Cura & then uninstalled an older version it breaks the links between Thingiverse & Cura. The solution is to do a reinstall of your current version of Cura, which restores the links

HellAholic, 2025-07-27:
The Thingiverse server is returning a HTML response instead of a model, from your logs. This might indicate an issue on the Thingiverse servers. So wait a while and see if it gets resolved by their team, otherwise raise an issue with the Thingiverse support with your specific case.

```
<!DOCTYPE html><html lang="en-US"><head><title>Just a moment...</title><meta http-equiv="Content-Type" content="text/html; charset=UTF-8"><meta http-equiv="X-UA-Compatible" content="IE=Edge"><meta name="robots" content="noindex,nofollow"><meta name="viewport" content="width=device-width,initial-scale=1"><style>*{box-sizing:border-box;margin:0;padding:0}html{line-height:1.15;-webkit-text-size-adjust:100%;color:#313131;font-family:system-ui,-apple-system,BlinkMacSystemFont,"Segoe UI",Roboto,"Helvetica Neue",Arial,"Noto Sans",sans-serif,"Apple Color Emoji","Segoe UI Emoji","Segoe UI Symbol","Noto Color Emoji"}body{display:flex;flex-direction:column;height:100vh;min-height:100vh}.main-content{margin:8rem auto;padding-left:1.5rem;max-width:60rem}@media (width <= 720px){.main-content{margin-top:4rem}}.h2{line-height:2.25rem;font-size:1.5rem;font-weight:500}@media (width <= 720px){.h2{line-height:1.5rem;font-size:1.25rem}}#challenge-error-text{background-image:url("data:image/svg+xml;base64,PHN2ZyB4bWxucz0iaHR0cDovL3d3dy53My5vcmcvMjAwMC9zdmciIHdpZHRoPSIzMiIgaGVpZ2h0PSIzMiIgZmlsbD0ibm9uZSI+PHBhdGggZmlsbD0iI0IyMEYwMyIgZD0iTTE2IDNhMTMgMTMgMCAxIDAgMTMgMTNBMTMuMDE1IDEzLjAxNSAwIDAgMCAxNiAzbTAgMjRhMTEgMTEgMCAxIDEgMTEtMTEgMTEuMDEgMTEuMDEgMCAwIDEtMTEgMTEiLz48cGF0aCBmaWxsPSIjQjIwRjAzIiBkPSJNMTcuMDM4IDE4LjYxNUgxNC44N0wxNC41NjMgOS41aDIuNzgzem0tMS4wODQgMS40MjdxLjY2IDAgMS4wNTcuMzg4LjQwNy4zODkuNDA3Ljk5NCAwIC41OTYtLjQwNy45ODQtLjM5Ny4zOS0xLjA1Ny4zODktLjY1IDAtMS4wNTYtLjM4OS0uMzk4LS4zODktLjM5OC0uOTg0IDAtLjU5Ny4zOTgtLjk4NS40MDYtLjM5NyAxLjA1Ni0uMzk3Ii8+PC9zdmc+");background-repeat:no-repeat;background-size:contain;padding-left:34px}@media (prefers-color-scheme: dark){body{background-color:#222;color:#d9d9d9}}</style><meta http-equiv="refresh" content="360"></head><body><div class="main-wrapper" role="main"><div class="main-content"><noscript><div class="h2"><span id="challenge-error-text">Enable JavaScript and cookies to continue</span></div></noscript></div></div><script>(function(){window._cf_chl_opt = {cvId: \'3\',cZone: \'www.thingiverse.com\',cType: \'managed\',cRay: \'965982e6d824e544\',cH: \'M6Pd2eKWaaECXc2jZX3ghcf1p3573oBKagHK0nj3kB0-1753591778-1.2.1.1-WKhjel28QafLkrWY4uNpCw0QlUv9ee99PyrD8pywfP_SHobiQ.i88fFrYQYV5G19\',cUPMDTk:"\\/download:12862488?__cf_chl_tk=8eMFJLBIYtsF6mjIs_05dTiujt2o0UqRAjnCmX.bXXg-1753591778-1.0.1.1-YhX4vnuMsTSQRGmQBSxxGUZjjT8nicGgDiihUBJE4Wg",cFPWv: \'b\',cITimeS: \'1753591778\',cTplC:0,cTplV:5,cTplB: \'cf\',fa:"\\/download:12862488?__cf_chl_f_tk=8eMFJLBIYtsF6mjIs_05dTiujt2o0UqRAjnCmX.bXXg-1753591778-1.0.1.1-YhX4vnuMsTSQRGmQBSxxGUZjjT8nicGgDiihUBJE4Wg",md: \'D8JeSrFXT9oiHNuJYPttYfzA063EVDFaX_.2d54QBb4-1753591778-1.2.1.1-oF.Qd09ZjulzJPpyNsEKdymqIvf3SMz8MaRnj4hNGozl8hmi6Com47589m4vWshnmKD0qAQMlTqstGsm9UAnL43.DhulUl5n_BLkaA1tWRWrtyySghQSM870WbdFtTOghzQTwlHj4kEYZx90Pf4Ht_I7js2mX1QYjOmpBpRo1QfDEorlzzjEt3aDw2ZUHuD_UHT8RTqMupyblloTVq1FZPvyQttVKNMLPwmf34mCKoRa6YsRtJ2c.EQlDVfDH6.L._R2xITys.1RY8KUZenHuAi4UjKkF1ewbSrhNHa1ntMrGLpRAiRHGHzw8njO.9a_3Ngql3a9rXiKNT_JprBO.QxGVRHsxDb2F62PzVK4VG9uxbOc9X.X9MEuyhi2WX8glqgFgs_uTOW2YhDWY5TkTnBVUj3i7kjDUGc8pOx01lYcISgzsmqY8Ptg33TMXEUblQH5K9ZzKkgygCU0nzWRY87viIolxXC6mUp6R129IEtVJQNz7dYXupmPT.L9.jSSVQXHigOSV4Vg9um.8ipzomIQ6JGvdN.LAHnl_4aQMU1.KZM9VI8_5rbvtgdJoMtIbv3i6kEmOlF7QwraH8SGgipj8f_7J56MIvdP5HvdtVXolLY8s2fDLlcyU7kb8BSGeJxjOeUIq6t3IwwAH_0Shln62dPoWmsoGhjauFNVSxdLCPIRqDhBkdZDZfV1d01gcY.vbyTcMkXUQDSaYVeS8bA1XIqOvVmCcASuicm7cbdLJdLNOteXzL1ABylHkjIJS4xCClu49v6nE8oJiGfhMsFkTiosMvKQvXChSU6QuVJFMpgKex4rhyL9O1l4TiryqqfwUzMNXofccONzu759IG8MYiE79XddsL7HsldC6.bzfqVcNQhRaiyMpquGyilCvOEbIq_WbFfu0.Ax_U6v.QMYFs_kLyY2hCOikmjukXUAf60dsyakCxIYnh7oPU3GBgA3yKO.j.zdGTowDr9G9A\',mdrd: \'EXE_H4TTar8Dfqrmokz7y_uKGfNqw6kCxdyMsZsb_mA-1753591778-1.2.1.1-yBSqJ_1Z1F6.F9PQ5ajMK986NqYTfm2j.cotpQ1GUwZGUkD37zQ8pHKPuwYa01Uwqc9YVbOl3gl2bfiSuAWN3HPcWZKRciRRRmCx0wdFMuSKZtMJ2bPUl.ajW8Tzkh_psAnrdeHgMlxa048xa4RZQjqFjsi6FHZZsdpjnwU6xz.EXEiKHAF0VkO59sWIlpR89TYXi0nfi5rgO6gcV2.2uGx77aSVsqPJgTBnEw1e42oIr0krTR.xDLuM6T4.Jz5C2juVatF8ig.jhWhnV.Dn0cysdFWilDfRsn60BaIdljYwkiNb2UMJXDmv7pY4q7V.DPs3zpEav8gsMPiOLf1y1L8CjMNYBrUHhAr0v4yD6bX4qSb.Pjm64MjYZN5FBhIOV7OiZ5AFJO2qgY30SxjUnq8ulA_Ec5Ky2N0PSp475FLoohMZb3hqub2DAueM43RCJMHUSTip0IQOH1dNivAJhEJm2Bt0yrpTg6fFTiY45SYSbXyH5So8mf1b7R4KErlMOrcb5nnLDcfEFwqW90lf2dG87rXGmIEquXqbduo1yLEToYk9HFkbVaxUqlgF9sCm3rFHwWnt.3Y.giSBLnFBVpeRmkl36CNd30xUvHFVRRWhWv_6Tb1bioTp0N5HFa6xHyyTb4JOriC6.MnbUvZ1KccQ4ok_1k6RiLYSNqw8QbHRxlkA_dzvWnYD5ek3KaFyaECZtIJ0XHMRgIPXUz9sWZG4vG5wwbevUbIKwVgaxwuzVEdAvZVw8_51bcWEVXZIOoilLq_kfzbrs7ZNajIySGRUHfwslEk0hfOgVkO4y5PgWZomGxXkP4up9vFl6IQYZDyaQ2olJ6N7gxL6rSjzbnxIk.BEPlaNVjrWtgvhsWD6uibkdMpqdoqquEP4LzgQ3eW3GTVqcIqoKeim0tX2IkieTciCofh.rmolkL68gY8xQVU4afQZm0BRraDF42kCx.C.BKGtNrPVusRIimo5fe.6JcDX_5TIF4QyJ1i6TorbcjsXYUKCA7Y8cwJWh5fOdYhTnj_VbGXNS6WhtD39OTqBe60lcKAMMZcM_YacZ9e0lSoW3.kcy1T.pgJqEYJV_O7ZSK8iYBDRoY4fBcjereKwZ4PN3a19FfDhKYoud4QSoB9mowJQX32Hk7Jb4dk3NF1RX9AI7LK2qGQRxNpaDLtRHLZt2x9PTGGlUzXXMqxDp2VH8TGdWQabFq5qK20N2bQJevMjdPzdge_ZReIIOcKNlB8KHKAtrYx6eehSi_TkqlEAjKStVMeiXig0RgZS0xvPbdD8KxbdLvmrMuhBHiXcx9t0_g51gWRLSVE04SF1dql5L7_mGDQhU7uNCcZStQwQOZ66SJq2cNvAfw9lIVElQHdOX4.0BkhJx1apMnDnJZBTedxk5QELKvOciMlVF_dq1K.RThNXdJlIjL7W58mEuOhZF8JG1IfTusKLy_AMVRy.JRtLf8.6vmeLJ6WLM8GFSahHSs0Sm [... cut ...] ch;if (window.history && window.history.replaceState) {var ogU = location.pathname + window._cf_chl_opt.cOgUQuery + window._cf_chl_opt.cOgUHash;history.replaceState(null, null,"\\/download:12862488?__cf_chl_rt_tk=8eMFJLBIYtsF6mjIs_05dTiujt2o0UqRAjnCmX.bXXg-1753591778-1.0.1.1-YhX4vnuMsTSQRGmQBSxxGUZjjT8nicGgDiihUBJE4Wg"+ window._cf_chl_opt.cOgUHash);a.onload = function() {history.replaceState(null, null, ogU);}}document.getElementsByTagName(\'head\')[0].appendChild(a);}());</script></body></html>

HellAholic, 2025-07-30:
The simplest possible recovery path would be clearing out the cache and configuration, this will return Cura to a fresh install state without any user profiles, added printers, or added plugins.
The installation does not change behavior as the time passes so if you're experiencing different issues that reach the level of being unable to open any files, then something is added in either of the paths below that is interfering with normal operations.

Note this will remove your printer configuration, plugins, or any profiles you have for your printer so Export the settings and print profiles that you have modified beforehand.

Close All Cura Instances.
For each of the paths, first make a back up, then delete all the folders (the name of the folders line up with Cura versions you previously or currently have installed).
- `%localappdata%/cura`
- `%appdata/cura%`
After that:
- Open Cura
- Add your printer
- Open a model 
- See if it opens the file
If not it might be a AV that is blocking operations

MandoJay, 2025-07-31:
I'm sorry, I don't know what that first step means (%localappdata%/cura). I tried putting that into the search bar and it just took me to a web search. I did google how to clear cache and configuration, and it had me go into the configuration folder through the Help menu in Cura. then close Cura, then delete the entire contents of that folder, leaving the folder itself (After that didn't work, I also deleted the entire folder & tried the steps again, neither worked.) After doing that, I restarted Cura and it had me add my printer again, as it did from the very first use. Did those steps essentially accomplish the same ones that you were advising? f so, that didn't work unfortunately. The files saved on my computer just keep asking me what program to open them with over & over (If it helps, their icons do not have the Cura logo next to them anymore & haven't since this problem began.) Trying to open something on thingiverse just gives me the Loading Model Failed message again. Opening files by using the "Open Files" option does seem to work, but won't allow me to edit the files any further if they've been sliced (although that may not be a bug but something that has always been the case) so I suppose that's a workaround, if all else fails..  If that wasn't what you were advising, please advise again and ELI5? Thank you in advance!

MandoJay, 2025-07-31:
Further information that may or may not be useful- When previously using Cura (before the steps tried today) supports were not showing in "Preview", & did not appear to be printing on the printer, despite having them enabled. After trying the steps I detailed & using the workaround of downloading the file from thingiverse and then using the "Open Files" option in Cura, the supports are showing in Preview now. Not sure if that helps at all.

HellAholic, 2025-07-31:
You can paste the `%localappdata%/cura` and `%appdata/cura%` in a file explorer on the top and it will directly take you to cache and configuration folders.

[image]

Unfortunately I don't see any clear signs to point you towards a solution with certainty, I can offer few starting points but as the issue is getting progressively worst on your system, I'm not sure if there is something that is caused from the Cura side.

Some potential starting points:

- Clear the cache and configuration folders (cache is more important), Uninstall Cura, then Install Cura again (Uninstall & install will refresh the file association without the need to manually edit the registry).
- I checked the open from Thingiverse on two different windows systems and they both worked, so could you try using another browser that doesn't have an ad-blocker? It might also be the permissions.
- Windows 11 is a bit restrictive when you open an application as administrator, if you are defaulting the cura to open as administrator that might be a contributing factor.
- You might not have enough space on the location that the "Open In Cura" option on TV first downloads the file to (usually temp folder on your operating system drive).

MandoJay, 2025-08-01:
Not really sure how this is a thingiverse issue when, again, Cura isn't working on files already saved to my computer previously and have nothing to do with thingiverse, but it's clear that this isn't getting resolved here. I will try with thingiverse, although sounds like it will be easier if I just find another slicer to use.

closed 2025-08-01 by MandoJay

MandoJay, 2025-08-01:
Thanks for the quick responses btw.