Ultimaker/Cura #20824: Error de posicionamiento a cargar proyecto guardado

CharlyUY, 2025-07-30:
### Cura Version

5.10.1

### Operating System

Windows 10

### Printer

Artillery Hornet

### Reproduction steps

1. Importo un modelo stl
2. bajo el modelo para que la mitad inferior quede debajo de la linea de la cama, para imprimir solo la mitad superior
3. realizo la segmentacion y la vista previa queda bien
4. guardo el proyecto
5. cierro el programa
6. vuelvo a abrir el programa
7. abro el proyecto guardado
8. ERROR: los objetos se mueven en el eje Z hacia arriba, como cuando recien son importados, no quedan cortando su parte inferior.

### Actual results

El proyecto abre con los objetos mal posicionados

### Expected results

El resultado esperado, es que al abrir un proyecto que ha sido guardado con los objetos movidos hacia abajo en el eje Z, se abra con los objetos en la misma posicion que han sido guardados

[image]

### Add your .zip and screenshots here ⬇️

[AH_Rueda - Adhesivo Plegadora.zip](https://github.com/user-attachments/files/21517719/AH_Rueda.-.Adhesivo.Plegadora.zip)

GregValiant, 2025-07-30:
Thanks for the report.
It appears that Cura is being fooled because you have 4 models, and then you "grouped" the models and appear to have lowered the "Group" to -8mm.  Since the 4 models are now a group, printing "One at a Time" will have no effect because Grouping has turned the 4 models into a single model (as far as Cura is concerned).

I think you can get what you want by changing the order you did things (changing the Z, and "Grouping").  There has been some other odd behavior noted recently when "Grouping" was involved.

What was the file format of the original model file (STL, 3mf, OBJ, etc.)?

@HellAholic may take a look.

HellAholic, 2025-08-14:
Thanks for the report.
We have a ticket for updating the 3MF opening logic and updating it to the newer method we use in UCP files, I added this issue as part of that ticket:

Internal reference:
CURA-12685