Ultimaker/Cura #20834: Wireless Connection

achillesmjb, 2025-08-02:
### Cura Version

5.10.1

### Operating System

Mac OS Latest version

### Printer

Ultimaker S3

### Reproduction steps

I added the IP address to CURA when it failed to see the printer. After that CURA connected to the printer but fails to connect to printer at later times. Constant battle to get connection. Definitely on same network. Printer indicates it is connected yet CURA often does not see it. 

### Actual results

As mentioned CURA often fails to connect wirelessly .

### Expected results

Constant problem connecting wirelessly.

### Add your .zip and screenshots here ⬇️

Cannot get wireless connection

GregValiant, 2025-08-02:
Thanks for the report.
This could well be a bug but I think you would get faster service with UM Support (or maybe on the ["UM Forum"](https://community.ultimaker.com/forum/106-ultimaker-3d-printers/)).  I don't know if anyone on the UM Support Team pays attention to these Github bug reports.

achillesmjb, 2025-08-02:
> GregValiant

Problem solved in a way. I have to go to manage printers in Cura, hit refresh, connect, then Cura connects so I’m satisfied.

Thank you.

Michael Brocoum