Ultimaker/Cura #20838: Main Interface is HUGE!

rebeltaz, 2025-08-04:
### Cura Version

5.10.2

### Operating System

Ubuntu 22.04 LTS

### Printer

irrelevant

### Reproduction steps

Open Cura. 

### Actual results

The main interface is HUGE. To the point that I cannot see the entire screen. In the Settings window, I cannot even see the Accept / Cancel buttons.

[image]
[image]

### Expected results

I would expect a usable sized interface. This is the main user interface window for the 4.8.0 version that I routinely use. 

[image]

As you can see, that looks fine. 

### Add your .zip and screenshots here ⬇️

I added the screenshots in the proper context dialogs.

HellAholic, 2025-08-04:
Current Cura version uses the operating system's parameters such as Font size and display scale to adjust the size of the UI components. I suspect your current setting for scale factor is above 150% based on the bottom bar's text font size.
Returning the scale factor to 100% should return the sizes to the expected values.

Alternatively, you can create a custom Cura theme for your operating system settings and place it in the configuration folder under themes. Example of a theme can be found here:
https://github.com/Ultimaker/Cura/tree/main/resources/themes/cura-light

The font and element sizes are in the theme.json file.

rebeltaz, 2025-08-08:
> Current Cura version uses the operating system's parameters such as Font size and display scale to adjust the size of the UI components. I suspect your current setting for scale factor is above 150% [...]

My scaling is at 1.00 (100%)

[image]

> Alternatively, you can create a custom Cura theme for your operating system settings and place it in the configuration folder under themes. Example of a theme can be found here: [...]

I will look into that and see how it goes. Thanks.

rebeltaz, 2025-08-08:
> I will look into that and see how it goes. Thanks.

Well... I copied all four of the themes into the .config directory. I noticed that the Dark theme inherits the text size from the Light theme, so I just edited that one and selected it instead, just for test purposes. Just to see the effects, I changed ALL of the font sizes to 0.55. When I reloaded Cura... there was NO difference in size...

[image]
[image]

🤷‍♂️

rebeltaz, 2025-08-09:
> Below the font section there is a sizes section as well, those control the UI element sizes. You would also need to give the theme an identifiable name to be able to distinguish it from the default one. Here is an example of the theme.json file for a custom theme that has 50% reduced sizes on most elements as reference.

Ahhh... I discovered my issue with the settings you suggested. I was under the impression that the configuration directory was ~/.config/cura/5.10 ... Apparently the ACTUAL configuration directory is ~/.local/share/cura/5.10/themes

The file you provided makes the UI a more manageable size. I'll work with it to create dark mode version. 

Thank you. I appreciate all of your help.

rebeltaz, 2025-08-09:
> [cura-custom.zip](https://github.com/user-attachments/files/21679781/cura-custom.zip)

I do have one question, though. What program did you use to edit the theme file? Every program I try to edit it in shows it as a single, one line, runon sentence instead of formatted text. Sorry to be a bother.

HellAholic, 2025-08-10:
I used visual studio code, it has extentions for different file types and automatically formats the file.