Ultimaker/Cura #20839: Cannot get past the "Add Printer" window when installing and trying to use Cura

deanocko, 2025-08-04:
### Cura Version

5.10.1

### Operating System

Windows 11

### Printer

Printer Agnostic

### Reproduction steps

I'm not sure how I initially encountered this.

### Actual results

I was having issues with my computer and deleted and then reinstalled Cura. Cura is now stuck on the "Add Printer" page and there is no way for me to get past it as shown in the image.

[image]

I work in a 3D printing lab at a university and we have 20 Ultimaker S5s which have been added to our account on Digital Factory. I have tried removing one and re-adding it to get past the dialog but it does not work. I have tried uninstalling and re-installing Cura multiple times. I have also tried repairing Cura. I have even tried past versions of the app and it still doesn't work.

Please fix ASAP as I need this app to assist students!

### Expected results

Can launch and use Cura as normal.

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2025-08-04:
Thanks for reaching out. Unfortunately, the information provided doesn’t give us enough detail to identify this as a Cura issue, and without reproduction steps, we can’t investigate it effectively.

Since this seems related to your setup and account configuration, I recommend contacting [Ultimaker Support](https://support.ultimaker.com/s/) directly as they might already have a known solution / workaround for it.

Potential workaround:
- Instead singing in during the welcome page, add a non-networked local printer - this can be a UM printer or a 3rd party one. That should get you past the initial page. Sign in after that when you are in the main Cura environment (printer added and buildplate visible).
- If you cannot add any printers (even local non-networked printers), include that information in the support request.
- Include the cura.log in your request, that is located under `%appdata%/cura/5.10`

bgaderlund, 2025-09-05:
Well, I think I found a way to get past screen.  I click Add local printer and just kept clicking through until the add printer screen went away. Would be really nice to have a way to close the splash screen!