Ultimaker/Cura #20844: Multi-Part Post Processing

marksy95, 2025-08-05:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Longer LK5 Pro

### Reproduction steps

A simple model (cube) is printed and paused at layer 73/78 using post-processing script. The pause is successful when printing a single cube. The pause is unsuccessful when printing multiple cubes (4).

Even when successful the pause does not wait for user input to resume.

### Actual results

1x - Print pauses, but resumes automatically (layer 74/78).
4x - Print does not pause.

### Expected results

Expected the printer to pause at layer 73/78. Also, the pause should wait for user input to resume.

### Add your .zip and screenshots here ⬇️

[LLK5PRO_Marker_4x_Pause.zip](https://github.com/user-attachments/files/21603759/LLK5PRO_Marker_4x_Pause.zip)

[image]

[image]

HellAholic, 2025-08-05:
Both Gcodes you shared contain the unconditional pause command `M0` for Marlin at layer 74 (Gcode layers starts with layer 0)
[image]

If the printer ignores the M0 Marlin command and resumes automatically on the single cube, then the problem is the machine not listening to M0 commands as it should pause the print and waits for user interaction on the LCD screen. 

There is no Pause SD print (M25 for pause and M24 for resume) so the printer should not resume after the pause without user input. You can try using the M1 command to see if that works but otherwise the issue seems to be on the printer firmware side.

GregValiant, 2025-08-06:
Regarding not actually pausing:
I'm seeing the same thing.  The "M0" (and "M1") command ("emergency stop") must be configured in the firmware.  If it doesn't work then you will need to try the other options.  M25 works for many printers.
(FYI - M0 works on my Ender and to resume there must be a button click on the LCD.)

If you are pausing for a filament change then maybe the post processor "Filament Change" would be a better choice.  It uses the "M600" command and puts all the settings onto the firmware so only a single line is inserted into the gcode.

I did the last revision of PauseAtHeight in 2022 and there have not been many bug reports about it.  Those reports that have come in are in regards to the proper pause command to use.  "Which command" is very machine dependent.

Regarding number of models:
The pause script doesn't care (nor does it have any idea) how many models are on the build plate.  It tracks the X, Y, E, Z locations until it finds the "layer of interest".  At that point it knows the settings, and the Gcode location, and it puts the text together with all the information and inserts it into the gcode.

I have a re-write of the script but I have not submitted it yet.  It will include "One at a Time" support.  The number of possible pause commands will become infinite because along with the 10 built-in command options it will have a "Custom Pause Command" box.  Regardless of that, _it will still be 100% dependent on the printer firmware_ reacting correctly to whatever pause command you use.

closed 2025-08-06 by GregValiant

marksy95, 2025-08-07:
Thanks for looking at the G-code for me. What's weird is the printer does pause for a little bit with the single part print (but automatically resumes, like I said) but doesn't pause at all for the 4-part.

What's even more odd is that I just hand a single-part print that didn't pause even those it's the exact same g-code I've used to print 13 of these already.

I will try the other M commands.

marksy95, 2025-08-07:
So, yes. M109 was being used and that's probably why it took so long to resume.

I have made my M104 the same temp as my M109 and replaced the M109 with M104.

So I've got

```
...
M104
M25
M104
...
```

I'm able to pause with the pause button and then the print resumes when I press the resume button. I have not tested to see how long I have to pause it before it continues on its own.

Thanks for all the help