Ultimaker/Cura #20854: Seam splitting

Kustomchoppers, 2025-08-09:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

I wonder if there is a way to have the seam staggered so that it is not a visible line running up the print.

### Describe alternatives you've considered

none

### Affected users and/or printers

sidewinder X1

HellAholic, 2025-08-09:
There is already a feature called Scarf Seam in the Experimental section.
In the Custom settings area search for Scarf Seam.

If you select a non-zero Scarf Seam Length, two other child settings are enabled:
[image]

You would need to experiment (hence the experimental setting) with the values and see what combination of settings for the Scarf seam works for your printer, material, and print settings.

Kustomchoppers, 2025-08-09:
Thank you, sorry for not finding that in the first place 🙂

Thanks, Daniel
Managing Director

www.kustomchoppers.co.uk<http://www.kustomchoppers.co.uk>
***@***.******@***.***>
01953 529150
07458149329 (WhatsApp)

image.png (view on web)<https://github.com/user-attachments/assets/62d9d856-b769-44be-a76a-4cb7c9b3e7e6>

closed 2025-08-09 by HellAholic