Ultimaker/Cura #20855: Slicing Problem

PWALSH2929, 2025-08-09:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ultimaker S7

### Reproduction steps

I put the stl files on the print bed

Spaced then out 

Proceeded to slice print job

### Actual results

However, I received an error message.

I just tried to save my file but its saying "layer height".

### Expected results

To give me the file to print my models

### Add your .zip and screenshots here ⬇️

Can't give you file nor screen shots

HellAholic, 2025-08-09:
Update to the latest version of cura and see if the issue is resolved. 5.4 version was released 2 years ago. Any fix to the issues is gonna be on the later builds.

https://github.com/Ultimaker/Cura/releases/tag/5.10.2

Without a project file or logs, this is the only solution available to offer.

PWALSH2929, 2025-08-09:
> image.png (view on web)
>
> If this is the issue you're experiencing. It's caused by the Plugins such
> as ElegooNeptuneThumbnails (version lower than 3.6.2)
> So updating the plugin would also resolve that issue.

So I have Cura 5.6.0-231299. I accidentally typed the wrong versoin. My
fault.

Asterchades, 2025-08-11:
> Cura 5.6.0-231299

This is a customised version of Cura provided by Elegoo. It's entirely possible they've included a faulty version of the plugin that HellAholic has referenced, in which case it would need to be updated. The question is whether it can be updated via the Marketplace, or if it's more thoroughly embedded in the client and requires an Elegoo-sourced update to correct it.