Ultimaker/Cura #20891: Unable to Reopen Settings Menu After Closing It

M-AN-D, 2025-08-24:
### Cura Version

5.10.2

### Operating System

macOS Sequoia 15.6.1

### Printer

Ender 3 pro + klipper

### Reproduction steps

Open Cura.
Navigate to the "Settings" menu.
Close the settings menu.
Attempt to reopen the settings menu.

### Actual results

The settings menu does not reopen, and the option to access it becomes unresponsive.

### Expected results

The settings menu should reopen when I try to access it again.

### Add your .zip and screenshots here ⬇️

[Schermopname 2025-08-24 om 20.51.50.mp4.zip](https://github.com/user-attachments/files/21958568/Schermopname.2025-08-24.om.20.51.50.mp4.zip)

M-AN-D, 2025-08-25:
Thank you for your response and suggestion!

I tested the situation again. After opening and closing the "Machine Settings" dialog, I found that the "Preferences" dialog indeed remains open and is hidden behind the main Cura window. Minimizing the main Cura window reveals the "Preferences" dialog, confirming the 'fix' you described!

HellAholic, 2025-08-25:
@GregValiant macOS window manager has a different behavior when the window is not modal/blocking. 
The fix is implemented and will be in the next release. 
https://github.com/Ultimaker/Cura/pull/20797
https://github.com/Ultimaker/Uranium/pull/1004

GregValiant, 2025-08-26:
@M-AN-D there you have the word from the man.  I'll go ahead and close this as your check is in the mail.

closed 2025-08-26 by GregValiant